openwrt/packages #3952: isc-dhcp: errors in startup script /etc/init.d/dhcpd6

pprindeville, 2017-02-03:
Maintainer: @antonyantony 
Environment: x86_64, generic, OpenWRT head
Description:

Looking at `/etc/init.d/dhcpd6` a few things stand out:

```
#!/bin/sh /etc/rc.common
 
START=65
 
lease_file=/var/dhcpd6.leases
config_file=/etc/dhcpd6.conf
pid_file=/var/run/dhcpd6.pid
 
start() {
        if [ ! -e $lease_file ]; then
                touch $lease_file
        fi
 
        /usr/sbin/dhcpd -q -6 -cf $config_file -lf $lease_file -pf $pid_file $dhcp_ifs
 
        if [ $? -ne 0 ]; then
                return 1
        fi
}
```

1. there is no '-6' argument you can pass to dhcpd: you run `/usr/sbin/dhcpd6 -q -cf ...` instead;
2. the variable `$dhcp_ifs` doesn't get set anywhere... not even clear where it should come from;
3. minor nit, but file existent (as opposed to path existence) is usually done by `-f` not `-e`.

I can fix all of this if I can figure out the answer to (2).

pprindeville, 2017-02-19:
@stintel, @dedeckeh

stintel, 2017-02-20:
@pprindeville please don't mention random people. Neither Hans, nor I have anything to do with this package. Hans doesn't even have write access to this repository.

I will not accept a PR without ACK from the package maintainer. I don't know what the policy is, in case a maintainer doesn't respond within a reasonable time. Maybe another thing discuss on the ML?

pprindeville, 2017-02-20:
Sorry if it wasn't clear: I was just hoping for some insightful comments.

dangowrt, 2017-04-14:
Apparently @antonyantony became unavailable or at least doesn't deal with PRs for this package within reasonable time. The same applies for #4237.
@openwrt/package-maintainers I suggest to make @pprindeville the new maintainer of this package, given that he's willing to do the job.
@pprindeville To agree please open a new PR changing the PKG_MAINTAINER line in
https://github.com/openwrt/packages/blob/master/net/isc-dhcp/Makefile#L17
We shall give @antonyantony another chance to react before merging that and hence officially reassigning maintainership.

dangowrt, 2017-06-05:
I believe this was solved in the meantime. @pprindeville also became the maintainer of isc-dhcp in the meantime, so he can sort out what ever else is still needed.

closed 2017-06-05 by dangowrt