openwrt/packages #3953: python-psycopg2 error build/compile (LEDE-trunk)

build000, 2017-02-03:
machine: WNDR4700 (apm821xx/powerpc_464fp)
make -j8 BUILD_LOG=1 **(error)**
```
dom@slawek:~/Pulpit/WNDR4700-LEDE-trunk$ scripts/getver.sh
r3286-c980147
```
cat/***/compile.txt:
```
running install
running build
running build_py
creating build
creating build/lib.linux2-2.7
creating build/lib.linux2-2.7/psycopg2
copying lib/extras.py -> build/lib.linux2-2.7/psycopg2
copying lib/pool.py -> build/lib.linux2-2.7/psycopg2
copying lib/__init__.py -> build/lib.linux2-2.7/psycopg2
copying lib/extensions.py -> build/lib.linux2-2.7/psycopg2
copying lib/psycopg1.py -> build/lib.linux2-2.7/psycopg2
copying lib/errorcodes.py -> build/lib.linux2-2.7/psycopg2
copying lib/tz.py -> build/lib.linux2-2.7/psycopg2
copying lib/_json.py -> build/lib.linux2-2.7/psycopg2
copying lib/_range.py -> build/lib.linux2-2.7/psycopg2
creating build/lib.linux2-2.7/psycopg2/tests
copying tests/test_module.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_dates.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/dbapi20.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_notify.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_errcodes.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/__init__.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_cancel.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_bugX000.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_quote.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_async.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_green.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_transaction.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/dbapi20_tpc.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_with.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_types_extras.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_cursor.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_extras_dictcursor.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_types_basic.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/testconfig.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/testutils.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_psycopg2_dbapi20.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_bug_gc.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_connection.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_lobject.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_copy.py -> build/lib.linux2-2.7/psycopg2/tests
warning: build_py: byte-compiling is disabled, skipping.

running build_ext
Error: pg_config executable not found.

Please add the directory containing pg_config to the PATH
or specify the full executable path with the option:

    python setup.py build_ext --pg-config /path/to/pg_config build ...

or with the pg_config option in 'setup.cfg'.
Makefile:55: recipe for target '/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/build_dir/target-powerpc_464fp_musl-1.1.16/psycopg2-2.6.2/.built' failed
make[3]: *** [/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/build_dir/target-powerpc_464fp_musl-1.1.16/psycopg2-2.6.2/.built] Error 1
```

commodo, 2017-02-04:
will take a look
maybe python-cryptography also needs some attention

banebg, 2017-02-04:
I also get error when i try build/compile python-setuptools, after update feeds some days ago:

https://github.com/openwrt/packages/issues/3947

commodo, 2017-02-04:
@banebg
will take a look at that too

build000, 2017-02-11:
problem fixed probably for this commit https://github.com/openwrt/packages/commit/4772aec581a2c161c804a558c262c067cf6d6f21
cat logs/***/compile.txt
```
running install
running build
running build_py
creating build
creating build/lib.linux2-2.7
creating build/lib.linux2-2.7/psycopg2
copying lib/extras.py -> build/lib.linux2-2.7/psycopg2
copying lib/pool.py -> build/lib.linux2-2.7/psycopg2
copying lib/__init__.py -> build/lib.linux2-2.7/psycopg2
copying lib/extensions.py -> build/lib.linux2-2.7/psycopg2
copying lib/psycopg1.py -> build/lib.linux2-2.7/psycopg2
copying lib/errorcodes.py -> build/lib.linux2-2.7/psycopg2
copying lib/tz.py -> build/lib.linux2-2.7/psycopg2
copying lib/_json.py -> build/lib.linux2-2.7/psycopg2
copying lib/_range.py -> build/lib.linux2-2.7/psycopg2
creating build/lib.linux2-2.7/psycopg2/tests
copying tests/test_module.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_dates.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/dbapi20.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_notify.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_errcodes.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/__init__.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_cancel.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_bugX000.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_quote.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_async.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_green.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_transaction.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/dbapi20_tpc.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_with.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_types_extras.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_cursor.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_extras_dictcursor.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_types_basic.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/testconfig.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/testutils.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_psycopg2_dbapi20.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_bug_gc.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_connection.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_lobject.py -> build/lib.linux2-2.7/psycopg2/tests
copying tests/test_copy.py -> build/lib.linux2-2.7/psycopg2/tests
warning: build_py: byte-compiling is disabled, skipping.

running build_ext
building 'psycopg2._psycopg' extension
creating build/temp.linux2-2.7
creating build/temp.linux2-2.7/psycopg
powerpc-openwrt-linux-musl-gcc -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -Os -pipe -mcpu=464fp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/dom/Pulpit/Projekty/WNDR4700-LEDE-17.01/build_dir/target-powerpc_464fp_musl-1.1.16/psycopg2-2.6.2:psycopg2-2.6.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-17.01/staging_dir/target-powerpc_464fp_musl-1.1.16/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-17.01/staging_dir/target-powerpc_464fp_musl-1.1.16/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-17.01/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.16/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-17.01/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.16/include/fortify -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-17.01/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.16/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-17.01/staging_dir/target-powerpc_464fp_musl-1.1.16/usr/include/python2.7 -fPIC -DPSYCOPG_DEFAULT_PYDATETIME=1 -DPSYCOPG_VERSION="2.6.2 (dt dec pq3 ext lo64)" -DPG_VERSION_HEX=0x090504 -DHAVE_LO64=1 -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-17.01/staging_dir/target-powerpc_464fp_musl-1.1.16/usr/include/python2.7 -I. -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-17.01/staging_dir/target-powerpc_464fp_musl-1.1.16/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-17.01/staging_dir/target-powerpc_464fp_musl-1.1.16/usr/include/postgresql/server -c psycopg/psycopgmodule.c -o build/temp.linux2-2.7/psycopg/psycopgmodule.o -Wdeclaration-after-statement
psycopg/psycopgmodule.c: In function 'init_psycopg':
psycopg/psycopgmodule.c:785:5: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     Py_TYPE(&connectionType) = &PyType_Type;
     ^
psycopg/psycopgmodule.c:788:5: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     Py_TYPE(&cursorType) = &PyType_Type;
     ^
psycopg/psycopgmodule.c:791:5: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     Py_TYPE(&typecastType) = &PyType_Type;
     ^
psycopg/psycopgmodule.c:794:5: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     Py_TYPE(&qstringType) = &PyType_Type;
     ^
psycopg/psycopgmodule.c:797:5: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     Py_TYPE(&binaryType) = &PyType_Type;
     ^
psycopg/psycopgmodule.c:800:5: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
[... cut ...]
rstrip.sh: /home/dom/Pulpit/Projekty/WNDR4700-LEDE-17.01/build_dir/target-powerpc_464fp_musl-1.1.16/psycopg2-2.6.2/ipkg-powerpc_464fp/python-psycopg2/usr/lib/python2.7/site-packages/psycopg2/_psycopg.so: shared object
Packaged contents of /home/dom/Pulpit/Projekty/WNDR4700-LEDE-17.01/build_dir/target-powerpc_464fp_musl-1.1.16/psycopg2-2.6.2/ipkg-powerpc_464fp/python-psycopg2 into /home/dom/Pulpit/Projekty/WNDR4700-LEDE-17.01/bin/packages/powerpc_464fp/packages/python-psycopg2_2.6.2-1_powerpc_464fp.ipk
copying: '/usr/lib/python2.7/site-packages'
```

build000, 2017-02-11:
issue to by close

closed 2017-02-11 by build000