openwrt/packages #3956: transmission-mbedtls fails to build

char11, 2017-02-04:
make[3]: Entering directory '/home/pio/1/openwrt/feeds/packages/net/transmission'
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /home/pio/1/openwrt/build_dir/target-mips_34kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git:transmission-2.92+git -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -std=gnu99  -I/home/pio/1/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/usr/include -I/home/pio/1/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/include -I/home/pio/1/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/usr/include -I/home/pio/1/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/include/fortify -I/home/pio/1/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /home/pio/1/openwrt/build_dir/target-mips_34kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git:transmission-2.92+git -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -std=gnu99  -I/home/pio/1/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/usr/include -I/home/pio/1/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/include -I/home/pio/1/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/usr/include -I/home/pio/1/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/include/fortify -I/home/pio/1/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/include " LDFLAGS="-L/home/pio/1/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/usr/lib -L/home/pio/1/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/lib -L/home/pio/1/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/usr/lib -L/home/pio/1/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/lib -znow -zrelro " make   -C /home/pio/1/openwrt/build_dir/target-mips_34kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git/. AR="mips-openwrt-linux-musl-gcc-ar" AS="mips-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap /home/pio/1/openwrt/build_dir/target-mips_34kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git:transmission-2.92+git -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -std=gnu99" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="mips-openwrt-linux-musl-gcc" GCC="mips-openwrt-linux-musl-gcc" CXX="mips-openwrt-linux-musl-g++" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-musl-objcopy OBJDUMP=mips-openwrt-linux-musl-objdump SIZE=mips-openwrt-linux-musl-size CROSS="mips-openwrt-linux-musl-" ARCH="mips" CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /home/pio/1/openwrt/build_dir/target-mips_34kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git:transmission-2.92+git -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -std=gnu99 -D_GNU_SOURCE" ;
make[4]: Entering directory '/home/pio/1/openwrt/build_dir/target-mips_34kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git'
Making all in extras
make[5]: Entering directory '/home/pio/1/openwrt/build_dir/target-mips_34kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git/extras'
make[5]: Nothing to be done for 'all'.
make[5]: Leaving directory '/home/pio/1/openwrt/build_dir/target-mips_34kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git/extras'
Making all in third-party
make[5]: Entering directory '/home/pio/1/openwrt/build_dir/target-mips_34kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git/third-party'
Making all in dht
make[6]: Entering directory '/home/pio/1/openwrt/build_dir/target-mips_34kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git/third-party/dht'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/pio/1/openwrt/build_dir/target-mips_34kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git/third-party/dht'
Making all in libb64
make[6]: Entering directory '/home/pio/1/openwrt/build_dir/target-mips_34kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git/third-party/libb64'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/pio/1/openwrt/build_dir/target-mips_34kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git/third-party/libb64'
Making all in libnatpmp
make[6]: Entering directory '/home/pio/1/openwrt/build_dir/target-mips_34kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git/third-party/libnatpmp'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/pio/1/openwrt/build_dir/target-mips_34kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git/third-party/libnatpmp'
Making all in miniupnpc
make[6]: Entering directory '/home/pio/1/openwrt/build_dir/target-mips_34kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git/third-party/miniupnpc'
make  all-am
make[7]: Entering directory '/home/pio/1/openwrt/build_dir/target-mips_34kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git/third-party/miniupnpc'
make[7]: Nothing to be done for 'all-am'.
make[7]: Leaving directory '/home/pio/1/openwrt/build_dir/target-mips_34kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git/third-party/miniupnpc'
make[6]: Leaving directory '/home/pio/1/openwrt/build_dir/target-mips_34kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git/third-party/miniupnpc'
Making all in libutp
make[6]: Entering directory '/home/pio/1/openwrt/build_dir/target-mips_34kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git/third-party/libutp'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/pio/1/openwrt/build_dir/target-mips_34kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git/third-party/libutp'
make[6]: Entering directory '/home/pio/1/openwrt/build_dir/target-mips_34kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git/third-party'
make[6]: Nothing to be done for 'all-am'.
make[6]: Leaving directory '/home/pio/1/openwrt/build_dir/target-mips_34kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git/third-party'
make[5]: Leaving directory '/home/pio/1/openwrt/build_dir/target-mips_34kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git/third-party'
Making all in libtransmission
make[5]: Entering directory '/home/pio/1/openwrt/build_dir/target-mips_34kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git/libtransmission'
  CCLD     bitfield-test
/home/pio/1/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/lib/gcc/mips-openwrt-linux-musl/5.3.0/../../../../mips-openwrt-linux-musl/bin/ld: copy reloc against protected `__stack_chk_guard' is dangerous
./libtransmission.a(crypto-utils-polarssl.o): In function `tr_rc4_new':
crypto-utils-polarssl.c:(.text+0x10c): undefined reference to `mbedtls_arc4_init'
./libtransmission.a(crypto-utils-polarssl.o): In function `tr_rc4_free':
crypto-utils-polarssl.c:(.text+0x11a): undefined reference to `mbedtls_arc4_free'
./libtransmission.a(crypto-utils-polarssl.o): In function `tr_rc4_set_key':
crypto-utils-polarssl.c:(.text+0x12e): undefined reference to `mbedtls_arc4_setup'
./libtransmission.a(crypto-utils-polarssl.o): In function `tr_rc4_process':
crypto-utils-polarssl.c:(.text+0x142): undefined reference to `mbedtls_arc4_crypt'
./libtransmission.a(crypto-utils-polarssl.o): In function `tr_dh_new':
crypto-utils-polarssl.c:(.text+0x15a): undefined reference to `mbedtls_dhm_init'
crypto-utils-polarssl.c:(.text+0x174): undefined reference to `mbedtls_dhm_free'
./libtransmission.a(crypto-utils-polarssl.o): In function `tr_dh_free':
crypto-utils-polarssl.c:(.text+0x19c): undefined reference to `mbedtls_dhm_free'
./libtransmission.a(crypto-utils-polarssl.o): In function `tr_dh_make_key':
crypto-utils-polarssl.c:(.text+0x1b8): undefined reference to `mbedtls_dhm_make_public'
./libtransmission.a(crypto-utils-polarssl.o): In function `tr_dh_agree':
[... cut ...]
make[3]: *** [Makefile:199: /home/pio/1/openwrt/build_dir/target-mips_34kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git/.built] Error 2
make[3]: Leaving directory '/home/pio/1/openwrt/feeds/packages/net/transmission'
make[2]: *** [package/Makefile:197: package/feeds/packages/transmission/compile] Error 2
make[2]: Leaving directory '/home/pio/1/openwrt'
make[1]: *** [package/Makefile:193: /home/pio/1/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/pio/1/openwrt'
make: *** [/home/pio/1/openwrt/include/toplevel.mk:194: world] Error 2

diizzyy, 2017-02-04:
Can you can the LEDE codebase a try instead?
https://github.com/lede-project/source

char11, 2017-02-05:
Build ok.

closed 2017-02-05 by char11