openwrt/packages #3959: ar71xx wdr3600 error make transmisson

pna1702, 2017-02-05:
config.status: creating Makefile
config.status: creating transmission-gtk.spec
config.status: creating cli/Makefile
config.status: creating daemon/Makefile
config.status: creating extras/Makefile
config.status: creating libtransmission/Makefile
config.status: creating utils/Makefile
config.status: creating third-party/Makefile
config.status: creating third-party/dht/Makefile
config.status: creating third-party/libb64/Makefile
config.status: creating third-party/libutp/Makefile
config.status: creating third-party/libnatpmp/Makefile
config.status: creating third-party/miniupnpc/Makefile
config.status: creating macosx/Makefile
config.status: creating gtk/Makefile
config.status: creating gtk/icons/Makefile
config.status: creating qt/config.pri
config.status: creating web/Makefile
config.status: creating web/images/Makefile
config.status: creating web/style/Makefile
config.status: creating web/style/jqueryui/Makefile
config.status: creating web/style/jqueryui/images/Makefile
config.status: creating web/style/transmission/Makefile
config.status: creating web/style/transmission/images/Makefile
config.status: creating web/style/transmission/images/buttons/Makefile
config.status: creating web/javascript/Makefile
config.status: creating web/javascript/jquery/Makefile
config.status: error: cannot find input file: `po/Makefile.in.in'
make[3]: *** [/home/user/source/build_dir/target-mips_24kc_musl-1.1.16/transmission-openssl/transmission-2.92+git/.configured_yyyyyyy] Error 1
make[3]: Leaving directory `/home/user/source/feeds/packages/net/transmission'
make[2]: *** [package/feeds/packages/transmission/compile] Error 2
make[2]: Leaving directory `/home/user/source'
make[1]: *** [/home/user/source/staging_dir/target-mips_24kc_musl-1.1.16/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/user/source'
make: *** [world] Error 2

Formating code blocks by wrapping them with pairs of ```
```

hnyman, 2017-02-05:
same issue that was discussed in https://bugs.lede-project.org/index.php?do=details&task_id=420

It is possible that transmission is missing a PKG_BUILD_DEPENDS dependency for gettext/glib2 tools, or alternatively some optional language components could be diasbled (like I tentatively proposed in FS#420)

@obsy

nidstigator, 2017-03-21:
i am having same issue when building for mvebu. Any idea what's the problem?

cfpp2p, 2017-03-22:
@nidstigator 
https://github.com/cfpp2p/transmission/issues/1#issuecomment-129141064
https://github.com/cfpp2p/transmission/issues/1#issuecomment-129548676  [re-generate Makefile.in.in]
https://github.com/cfpp2p/transmission/issues/1#issuecomment-129575933  [intltoolize]

alive4ever, 2017-03-23:
Upstream [suggest](https://github.com/transmission/transmission/blob/master/README.md) building with ``cmake``.

neheb, 2018-03-01:
CMake is stalled on Travis being broken.

diizzyy, 2019-03-28:
No issues compiling transmission using master branch reported by buildbots, closing.

closed 2019-03-28 by diizzyy