openwrt/packages #3966: Python_Setuptools 2.7.13 is an outdated package

DazzyWalkman, 2017-02-07:
Package python-setuptools version has been changed from 27.2.0 to 2.7.13 since commit "pip,setuptools: move as part of python,python3 build"
https://github.com/openwrt/packages/commit/a53d0c5a403d1669e2cf6c59c2be6a9d3ed633a0
Please note that python setuptools uses a different version numbering scheme from python main package. So, following python 2.7 current version number, setuptools 2.7.13 is a dated release.
I tried to install python package  html2text [https://pypi.python.org/pypi/html2text/2016.9.19]() under setuptools 2.7.13 using python setup.py build/install method, and got a couple of warnings like 

> /usr/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'entry_points'

 , then installation failed.
There's not any problem under former setuptools 27.2.0.
`pip install --upgrade setuptools`
fixed this issue.

hnyman, 2017-02-07:
@commodo

commodo, 2017-02-07:
@DazzyWalkman 

hmm ; so, Python setuptools is actually 28.x.y  something ; 
the thing is, that I've changed the setuptools & pip versions to the ones that come packaged with Python & Python3 (which are Python setuptools 28.x.y and Pip 9.y.z ]
It's less maintenance work for me [ 6 packages into 2 ]

but good point: for pip & setuptools, I'll try to extract the correct PKG_VERSION
@hnyman any hints on how to override sub-package version ?

hnyman, 2017-02-07:
> any hints on how to override sub-package version ?

I think that you can just define the version in python-package-setuptools.mk where your define the setuptools package

There have been some packages that have defined different versions for multiple packages in the same Makefile.

One example is the old deprecated usb-modeswitch (and -data). There the main package usb-modeswitch has PKG_VERSION:=1.2.5 while subpackage usb-modeswitch-data has VERSION:=20121109
https://git.lede-project.org/?p=source.git;a=blob;f=package/system/utils/usb-modeswitch/Makefile;h=d82a83c56a1709278bd49004e86bd18f37ca3e10;hb=041ac5dbff0ba3c782572000c3fa7de73971f315

(slightly older) end result is still visible in AA12.09 download dir:
usb-modeswitch-data_20120120-1_ar71xx.ipk
usb-modeswitch_1.2.3-2_ar71xx.ipk

commodo, 2017-02-07:
cool ; will try that 
thanks

closed 2017-02-09 by hnyman