openwrt/packages #3969: ddns-scripts: DUIADNS update-url is old, needs updating

freegigi, 2017-02-08:
Maintainer: @chris5560 
Environment: 	TP-Link TL-WR1043N/ND v2, LEDE SNAPSHOT

Description:
The update url for DUIADNS service is old. It should be something like: 
```
http://ip.duiadns.net/dynamic.duia?host=[DOMAIN]&password=[PASSWORD]&ip=[IP]
```
(source: my duia admin interface)

Thank you.

chris5560, 2017-02-08:
@freegigi
Thanks for pointing out.
Could you please also provide information about IPv6 update url.
So I can update both services files.
Many thanks

freegigi, 2017-02-08:
The admin interface sais that the update url is (replacing domanin and pwd):

`https://ip.duiadns.net/dynamic.duia?host=[DOMAIN]&password=[PASSWORD]&ip4=[ipv4_ip]&ip6=[ipv6_ip]`

I tested and if only one parameter is present the update works ok, http or https.

In a blog post, the developer said that one can use 'ip' parameter for any type of address, ipv4 or ipv6, so I used that in my first post. It works, I tested it. But probably it's safer to use ip4 and ip6, because that's what the admin interface says and looks cleaner.

Also, in the password field one can use cleartext or md5sum (their own client uses by default https and md5 password). Maybe ddns-script can use md5 too, it's a nice feature.

Hope this helps.

chris5560, 2017-02-09:
@freegigi

Thanks for feedback.
Updates prepared.
md5sum for duiadns.net only is to much overhead because you need to handle duiadns.net separately.
Have you ever log the "answer" you get from duiadns.net on successful update ?
I didn't find any information on that at "google".
So I can extend services files on next update.

Many thanks

freegigi, 2017-02-09:
No logs, sorry.

Not sure what you need, but maybe this helps - I used curl to get the server message:

On successful update (same even if new or the same ip is used):
```
<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.01//EN">
<html><head><title>DUIA Module</title></head><body><ul><li>Hostname myhost.duia.ro</li><ul><li>Ipv4 aaa.bbb.ccc.ddd</li></ul></ul></body></html>
```

Wrong password:
```
<HTML><HEAD>
<TITLE>500 Internal Server Error</TITLE>
</HEAD><BODY>
<H1>Internal Server Error</H1>
</BODY></HTML>
```

closed 2018-08-17 by dibdot