openwrt/packages #3977: ffmpeg: configure error in 3.2.2 update (reprise)

artynet, 2017-02-10:
TARGET : ar71xx
musl : v1.1.16
arch : mips

hello @thess,

bringing up again this issue :

```
getting configure error for libmp3lame

ERROR: libmp3lame >= 3.98.3 not found

please fix this...
```

since I forgot to answer here https://github.com/openwrt/packages/issues/3836. The problem still persists in the LEDE and OpenWRT trunks as well, attaching the configure log as requested.

[ffmpeg-3.2.2-config.log.txt](https://github.com/openwrt/packages/files/767022/ffmpeg-3.2.2-config.log.txt)

Lame 3.99.5 is installed since required as prerequisite for **libffmpeg-full** hope this helps....please let me know about that

Best, Arturo

thess, 2017-02-11:
Not sure what the actual problem is yet, but from your config output the ld errors are usually from a mix of incompatible ABI selections such as hard and soft float options (`ld: failed to merge target specific data...`).

I also see a configure arg `--enable-nonfree` which is not part of the 3.2.2 Makefile in our repo! I can only conclude you have a very non-standard environment and you need to supply more info. The output from:
> ./scripts/diffconfig.sh

may help. Try `make dirclean` to rebuild toolchain to see if that helps.

artynet, 2017-02-12:
@thess I added the configure option `--enable-nonfree` just for a simple attempt if I actually had to activate the nonfree features. I'll let you know and then report here....thanks !

artynet, 2017-02-17:
so @thess, the issue is still persisting with the latest fresh pull of LEDE :+1: 

```
 cd /home/arturo/Downloads/lede-source/build_dir/target-mips_24kc_musl/ffmpeg-full/ffmpeg-3.2.2; CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/arturo/Downloads/lede-source/build_dir/target-mips_24kc_musl/ffmpeg-full/ffmpeg-3.2.2:ffmpeg-3.2.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/arturo/Downloads/lede-source/staging_dir/target-mips_24kc_musl/usr/include -I/home/arturo/Downloads/lede-source/staging_dir/target-mips_24kc_musl/include -I/home/arturo/Downloads/lede-source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/usr/include -I/home/arturo/Downloads/lede-source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/include/fortify -I/home/arturo/Downloads/lede-source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/include -fpic" LDFLAGS="-L/home/arturo/Downloads/lede-source/staging_dir/target-mips_24kc_musl/usr/lib -L/home/arturo/Downloads/lede-source/staging_dir/target-mips_24kc_musl/lib -L/home/arturo/Downloads/lede-source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/usr/lib -L/home/arturo/Downloads/lede-source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/lib -znow -zrelro" ./configure --enable-cross-compile --cross-prefix="mips-openwrt-linux-musl-" --arch="mips" --target-os=linux --prefix="/usr" --pkg-config="pkg-config" --enable-shared --enable-static --enable-small --enable-pthreads --enable-zlib --disable-doc --disable-debug --enable-gpl --enable-version3 --disable-dxva2 --disable-lzma --disable-vaapi --disable-vda --disable-vdpau --disable-outdevs --disable-altivec --disable-vsx --disable-power8 --disable-amd3dnow --disable-amd3dnowext --disable-mmx --disable-mmxext --disable-sse --disable-sse2 --disable-sse3 --disable-ssse3 --disable-sse4 --disable-sse42 --disable-avx --disable-xop --disable-fma3 --disable-fma4 --disable-avx2 --disable-aesni --disable-armv5te --disable-armv6 --disable-armv6t2 --disable-inline-asm --disable-mipsdsp --disable-mipsdspr2 --disable-mipsfpu --disable-msa --disable-mmi --disable-fast-unaligned --disable-runtime-cpudetect --disable-yasm --enable-libopus --enable-decoder=libopus   )
ERROR: opus not found using pkg-config

If you think configure made a mistake, make sure you are using the latest
version from Git.  If the latest version fails, report the problem to the
ffmpeg-user@ffmpeg.org mailing list or IRC #ffmpeg on irc.freenode.net.
Include the log file "config.log" produced by configure as this will help
solve the problem.
Makefile:637: recipe for target '/home/arturo/Downloads/lede-source/build_dir/target-mips_24kc_musl/ffmpeg-full/ffmpeg-3.2.2/.configured_yyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyynynnnnyyynnynynnnynnynnnnnyyyyyyyyyyyyyyyyyyyyyynyyyyyyyyyyyyyyyyyyynnnnnnn' failed
make[2]: *** [/home/arturo/Downloads/lede-source/build_dir/target-mips_24kc_musl/ffmpeg-full/ffmpeg-3.2.2/.configured_yyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyyynynnnnyyynnynynnnynnynnnnnyyyyyyyyyyyyyyyyyyyyyynyyyyyyyyyyyyyyyyyyynnnnnnn] Error 1
make[2]: Leaving directory '/home/arturo/Downloads/lede-source/feeds/packages/multimedia/ffmpeg'
package/Makefile:105: recipe for target 'package/feeds/packages/ffmpeg/compile' failed
make[1]: *** [package/feeds/packages/ffmpeg/compile] Error 2
make[1]: Leaving directory '/home/arturo/Downloads/lede-source'
Build failed - please re-run with -j1 to see the real error message
/home/arturo/Downloads/lede-source/include/toplevel.mk:197: recipe for target 'package/ffmpeg/compile' failed
make: *** [package/ffmpeg/compile] Error 1
```

and still a pkg-config error....any thoughts about that ? thank you in advance...

thess, 2017-02-17:
I cannot help you if do not supply the requested info on your environment. This problem seems to be unique to you.

* Under what OS/system/platform are you doing your builds?
* Please attached diffconfig.sh output I requested
* Have your tried `make dirclean` ? Or, at least do something like `rm -rf tmp` before doing a re-build.

artynet, 2017-02-18:
hello @thess 

1. The host machine is based on Ubuntu Linux 14.04 LTS or 16.10 x86_64. The target arch is ar71xx mips BE 32-bit. I have disabled the option to build with MIPS16 instructions
2. no dramatic changes since I start from scratch usually
3. applies as above since I usually run : `make tools/install V=99 && make toolchain/install V=99`. Then I run `make menuconfig`, select the ffmpeg entry,  save and exit and finally run its build which fails with the given error...

artynet, 2017-02-18:
@thess great news ! I reinstated the :

`CONFIG_USE_MIPS16=y`

option and the build started and finished as it should....I think I have spotted the actual issue. It is usually an option a disable so it's good to know that it affects the package build !

thess, 2017-02-19:
Thanks for the update - can we close this now?

artynet, 2017-02-20:
hello @thess,

the build has successfully been tested with all the versions of libffmpeg including the custom one with all the options enabled ! I definitely think that we can close this issue. Thank you very much again for all of your support !

Regards, Arturo

closed 2017-02-20 by thess