openwrt/packages #3979: asterisk13: Build fails with SRTP (lede)

schoerg, 2017-02-10:
I used the latest Lede build from today.

Target: Allwinner A1x/A20/A3x
Target Profile: Bananapi

`make V=s`

Output:

`mkdir -p /mnt/hdd/lede2/source/bin/targets/sunxi/generic/packages /mnt/hdd/lede2/source/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.15_eabi/asterisk-13.9.1/ipkg-arm_cortex-a8_vfpv3/asterisk13-res-srtp/CONTROL /mnt/hdd/lede2/source/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.15_eabi/pkginfo
install -d -m0755 /mnt/hdd/lede2/source/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.15_eabi/asterisk-13.9.1/ipkg-arm_cortex-a8_vfpv3/asterisk13-res-srtp/usr/lib/asterisk/modules
install -m0755 /mnt/hdd/lede2/source/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.15_eabi/asterisk-13.9.1/ipkg-install/usr/lib/asterisk/modules/*res_srtp.so* /mnt/hdd/lede2/source/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.15_eabi/asterisk-13.9.1/ipkg-arm_cortex-a8_vfpv3/asterisk13-res-srtp/usr/lib/asterisk/modules/;
install: cannot stat '/mnt/hdd/lede2/source/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.15_eabi/asterisk-13.9.1/ipkg-install/usr/lib/asterisk/modules/*res_srtp.so*': No such file or directory
Makefile:404: recipe for target '/mnt/hdd/lede2/source/bin/packages/arm_cortex-a8_vfpv3/telephony/asterisk13-res-srtp_13.9.1-1_arm_cortex-a8_vfpv3.ipk' failed
make[3]: *** [/mnt/hdd/lede2/source/bin/packages/arm_cortex-a8_vfpv3/telephony/asterisk13-res-srtp_13.9.1-1_arm_cortex-a8_vfpv3.ipk] Error 1
make[3]: Leaving directory '/mnt/hdd/lede2/source/feeds/telephony/net/asterisk-13.x'
package/Makefile:133: recipe for target 'package/feeds/telephony/asterisk-13.x/compile' failed
make[2]: *** [package/feeds/telephony/asterisk-13.x/compile] Error 2
make[2]: Leaving directory '/mnt/hdd/lede2/source'
package/Makefile:129: recipe for target '/mnt/hdd/lede2/source/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.15_eabi/stamp/.package_compile' failed
make[1]: *** [/mnt/hdd/lede2/source/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.15_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/mnt/hdd/lede2/source'
/mnt/hdd/lede2/source/include/toplevel.mk:197: recipe for target 'world' failed
make: *** [world] Error 2
`

libsrtp is included in menuconfig.

The same applies with asterisk11.

I opened the issue on openwrt/telephony, but no response.

hnyman, 2017-02-10:
Asterisk lives in the telephony repo, so you might get more attention there:
https://github.com/openwrt/telephony

You apparently already have a ticket there
https://github.com/openwrt/telephony/issues/128

schoerg, 2017-02-10:
I know. Since I did not get any response there I opened it here too.

hnyman, 2017-02-10:
The Telephony repo is not that well maintained (the last commit there is from 15 Jul 2016), but your best bet might still be to target the people having committed recently to Asterisk with @ 

@jslachta  and @dangowrt seem to have made the last commits to asterisk13

schoerg, 2017-03-27:
At least asterisk11 seems to build now without problems (lede v17.01.0)

closed 2017-03-27 by schoerg