openwrt/packages #3981: perl 5.22.1 : HOST build fails with LEDE trunk

artynet, 2017-02-11:
I have recently bumped into this issue while building the latest trunk. Basically since the executable in 👍 

`staging_dir/hostpkg/usr/bin`

is not found, I have force the copy of it in the `Host/Install` function :

```patch
diff --git a/lang/perl/Makefile b/lang/perl/Makefile
index d3cb081..63325e9 100644
--- a/lang/perl/Makefile
+++ b/lang/perl/Makefile
@@ -84,6 +84,7 @@ define Host/Install
        ( cd $(HOST_BUILD_DIR); ./miniperl installperl )
        $(INSTALL_DIR) $(HOST_PERL_PREFIX)/bin/
        $(CP) $(HOST_BUILD_DIR)/generate_uudmap $(HOST_PERL_PREFIX)/bin/
+       $(CP) $(HOST_BUILD_DIR)/perl $(HOST_PERL_PREFIX)/bin/

        # Link any possibly installed static extension in
        $(MAKE) -C $(HOST_BUILD_DIR)/relink clean || true
```

does it work for you ? Greets.

diizzyy, 2018-06-11:
@artynet 
Still an issue?

diizzyy, 2019-03-28:
Closing due to timeout

closed 2019-03-28 by diizzyy