openwrt/packages #3984: php-7.1.1 build/compile error (LEDE-trunk)

build000, 2017-02-11:
@mhei 
```
dom@slawek:~/Pulpit/WNDR4700-LEDE-trunk$ scripts/getver.sh
r3386-0799de6
```
make -j8 **(error)**
make -j1 V=s
```
(...)
make[3]: Entering directory '/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/feeds/packages/lang/php7'
rm -f /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/build_dir/target-powerpc_464fp_musl/php-7.1.1/.built
touch /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/build_dir/target-powerpc_464fp_musl/php-7.1.1/.built_check
CFLAGS="-Os -pipe -mcpu=464fp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/build_dir/target-powerpc_464fp_musl/php-7.1.1:php-7.1.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/lib/libiconv-stub/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/lib/libintl-stub/include  -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl/include/fortify -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/lib/libiconv-stub/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/lib/libintl-stub/include " CXXFLAGS="-Os -pipe -mcpu=464fp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/build_dir/target-powerpc_464fp_musl/php-7.1.1:php-7.1.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/lib/libiconv-stub/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/lib/libintl-stub/include  -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl/include/fortify -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/lib/libiconv-stub/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/lib/libintl-stub/include " LDFLAGS="-L/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/lib -L/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/lib -L/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl/usr/lib -L/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl/lib -znow -zrelro -L/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/lib/libiconv-stub/lib -Wl,-rpath-link=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/lib/libiconv-stub/lib -L/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/lib/libintl-stub/lib -Wl,-rpath-link=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/lib/libintl-stub/lib " make  -C /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/build_dir/target-powerpc_464fp_musl/php-7.1.1/. AR="powerpc-openwrt-linux-musl-gcc-ar" AS="powerpc-openwrt-linux-musl-gcc -c -Os -pipe -mcpu=464fp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/build_dir/target-powerpc_464fp_musl/php-7.1.1:php-7.1.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/lib/libiconv-stub/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/lib/libintl-stub/include" LD=powerpc-openwrt-linux-musl-ld NM="powerpc-openwrt-linux-musl-gcc-nm" CC="powerpc-openwrt-linux-musl-gcc" GCC="powerpc-openwrt-linux-musl-gcc" CXX="powerpc-openwrt-linux-musl-g++" RANLIB="powerpc-openwrt-linux-musl-gcc-ranlib" STRIP=powerpc-openwrt-linux-musl-strip OBJCOPY=powerpc-openwrt-linux-musl-objcopy OBJDUMP=powerpc-openwrt-linux-musl-objdump SIZE=powerpc-openwrt-linux-musl-size CROSS="powerpc-openwrt-linux-musl-" ARCH="powerpc" ;
make[4]: Entering directory '/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/build_dir/target-powerpc_464fp_musl/php-7.1.1'
/bin/bash /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/build_dir/target-powerpc_464fp_musl/php-7.1.1/libtool --silent --preserve-dup-deps --mode=link powerpc-openwrt-linux-musl-gcc -DPHP_ATOM_INC -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/build_dir/target-powerpc_464fp_musl/php-7.1.1/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/build_dir/target-powerpc_464fp_musl/php-7.1.1/main -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/build_dir/target-powerpc_464fp_musl/php-7.1.1 -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/build_dir/target-powerpc_464fp_musl/php-7.1.1/ext/date/lib -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/include/libxml2 -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/lib/libintl-full/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/lib/libiconv-stub/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/build_dir/target-powerpc_464fp_musl/php-7.1.1/ext/mbstring/oniguruma -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/build_dir/target-powerpc_464fp_musl/php-7.1.1/ext/mbstring/libmbfl -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/build_dir/target-powerpc_464fp_musl/php-7.1.1/ext/mbstring/libmbfl/mbfl -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/include/mysql -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/build_dir/target-powerpc_464fp_musl/php-7.1.1/ext/zip/lib -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/build_dir/target-powerpc_464fp_musl/php-7.1.1/TSRM -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/build_dir/target-powerpc_464fp_musl/php-7.1.1/Zend  -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl/include/fortify -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/lib/libiconv-stub/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/lib/libintl-stub/include  -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/lib/libiconv-stub/include -Os -pipe -mcpu=464fp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/build_dir/target-powerpc_464fp_musl/php-7.1.1:php-7.1.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/usr/lib/libiconv-stub/include -I/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/ [... cut ...]
package/Makefile:101: recipe for target '/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/stamp/.package_compile' failed
make[1]: *** [/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk'
/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/include/toplevel.mk:197: polecenia dla obiektu 'world' nie powiodły się
make: *** [world] Błąd 2
```

mhei, 2017-02-12:
Currently, traveling. Need a few days before I can have a look...

artynet, 2017-02-20:
it's like are using the system-wide headers to build and not the local ones....

build000, 2017-02-21:
@artynet 
previously from the same sources being built normally - based on the headlines of the toolchain/target - perhaps the introduction of new solutions, "host-pkg"(etc.) disrupts this process and some packages do not want to build as before - you may need some new/modified patches/changes in the Makefile in some packages ...

artynet, 2017-02-23:
@mhei @build000 

the issue might be related just to the ```intl``` module. I have disabled it and the build ends as it should. I think a fixed path should be set as for libintl-full...

mhei, 2017-02-24:
Lets see, whether #4066 already solved this issue. At least the descriptions reads so...

artynet, 2017-02-24:
@mhei 
 
ok great, it seems to be fixed now and I also think we can upgrade to v7.1.2 as well. This morning I tried to figure out the problem working right the way of @nxhack passing as additional argument :

```--with-icu-dir=$(STAGING_DIR)/usr```

but `icu-config` was not found since not installed. Good Job !

mhei, 2017-02-24:
Yes, I'm still re-checking on my local machine with latest LEDE master and packages master. Upgrade to 7.1.2 is already commited locally, if everthing works fine, I'll push it out in a hour or so...

mhei, 2017-02-24:
Closing this issue, fixed by #4066.

closed 2017-02-24 by mhei