openwrt/packages #4006: ddns-scripts: password info stored in cleartext in log-files

AlfonsHeyrman, 2017-02-14:
Description
========
DynDNS stores passwords in cleartext in its log files. From a security point-of-view, it is considered bad practice to store sensitive information like passwords in cleartext format in log-files.

Details
=====
root@mymodem:/etc/config# cd /var/log/

root@mymodem:/tmp/log# cd ddns/

root@mymodem:/tmp/log/ddns# ls
myddns_ipv4.log  myddns_ipv6.log

root@mymodem:/tmp/log/ddns# cat myddns_ipv4.log 
 151617       : ************ ************** ************** **************
 151617  note : PID '9528' started at 2017-02-07 15:16
 151617       : ddns version  : 2.7.5
 151617       : uci configuration:
...
ddns.myddns_ipv4.password='testpwd'
ddns.myddns_ipv4.service_name='dyndns.com'
...

root@mymodem:/tmp/log/ddns# cat myddns_ipv6.log
...
ddns.myddns_ipv6.password='your_password'
...

hnyman, 2017-02-14:
@chris5560

chris5560, 2017-02-15:
DDNS user:password is also stored in cleartext in config file /etc/config/ddns
So why not inside log file ?
You can disable log by setting option use_logfile '0'
user:password used by ddns-scripts are not mission critical like user:password inside /etc/passwd
Other DDNS clients are also storing user:password in cleartext inside config and log

AlfonsHeyrman, 2017-02-17:
Hi Christian,

thanks for the fast reply.

An important problem is that the cleartext DynDNS credentials will e.g. be exposed via Syslog in case a Syslog server config is applied in OpenWRT/Lede. Customers of us do not expect sensitive info to be part of this Syslog exchange.

We believe the 'OpenWRT/Lede community' is sensitive to security, which e.g. is also reflected in the below commits which have been applied in the past.

- https://git.lede-project.org/?p=source.git;a=commit;h=96f0bbe91d2176284324c4d9b4839b6878148602
- https://git.lede-project.org/?p=source.git;a=commit;h=03cd41679597364ca0b382646b35cb5d06465d1b

chris5560, 2017-02-17:
Hi @AlfonsHeyrman

"debug" information are only logged to logfile but not to syslog.
Inside logfile you see "debug" logging. "[TIMESTAMP][BLANK]: ...."
All other are marked with their level "[TIMESTAMP][LEVEL]: ...." i.e. info warn etc.
For the next release I'm planing to have a separate "debug" option for logfile output.
In the past it was very useful to have full debug information inside logfile for support on errors.

AlfonsHeyrman, 2017-02-27:
Thanks Christian. My excuses for the wrong link to Syslog. The remaining issue is that we have use cases were a limited Cli is offered, allowing access to e.g. log-files. Not all the Cli users will have the rights to see the DynDNS pwd in cleartext. Would the above proposal allow for a filtering option, ensuring cleartext DynDNS pwds would not be shown in cleartext?

dibdot, 2018-08-17:
I think that has been fixed already - close this by now.

closed 2018-08-17 by dibdot