openwrt/packages #4010: cryptsetup: update to 1.7.3

blackfuel, 2017-02-15:
Maintainer: @dangowrt 

Please update to Cryptsetup 1.7.3.  It runs fine on my Asus RT-AC68U (ARMv7) router.  Thank you.
```
Cryptsetup 1.7.3 Release Notes
==============================

Changes since version 1.7.2

* Fix device access to hash offsets located beyond the 2GB device boundary in veritysetup.

* Set configured (compile-time) default iteration time for devices created directly through
  libcryptsetup (default was hardcoded 1 second, the configured value applied only
  for cryptsetup application).

* Fix PBKDF2 benchmark to not double iteration count for specific corner case.
  If the measurement function returns exactly 500 ms, the iteration calculation loop
  doubled iteration count but instead of repeating measurement it used this value directly.

* OpenSSL backend: fix memory leak if hash context was repeatedly reused.

* OpenSSL backend: add support for OpenSSL 1.1.0.

* Fix several minor spelling errors.
```

diizzyy, 2017-06-02:
@blackfuel
Update notifications are being sent via uscan runs by @sdwalker, no need to create issues for update requests. You are free to help by submitting a pull request of course. Please close this.

diizzyy, 2017-06-24:
@hnyman @thess 
Please close this

dangowrt, 2017-06-24:
Sorry, I wasn't aware of the ticket. And yes, I do receive @sdwalker notifications about upstream version updates.

closed 2017-06-24 by dangowrt