openwrt/packages #4014: RFC: move luci-app-sqm from packages to luci

danrl, 2017-02-15:
Maintainer: @tohojo
Environment: LEDE, latest, all platforms

Description:
I would like to move` luci-app-sqm`, which is currently defined in `net/sqm-scripts/Makefile` to the luci repository. As I understand, the lua source files are part of @tohojo's upstream repository of sqm-scripts. So I seek advice from @tohojo about objections to this move, ideas how to pull the lua files during the build process and maybe opinion on migrating the lua source files into the luci repository alltogether*. Other dev's opinions are also welcome. Thank you!

It is the last luci-app-* package that is part of packages from what I know. Other luci-app-* packages that are part of packages are in the process of moving (well, after related PRs have been discussed and reviewed of course)

*This means giving up some control over the files, though.

Related:
- https://github.com/openwrt/packages/pull/4013
- https://github.com/openwrt/luci/pull/1017
- https://github.com/openwrt/luci/pull/1018
- https://github.com/openwrt/luci/pull/1019
- https://github.com/openwrt/luci/pull/1020

tohojo, 2017-02-15:
Hmm, while it's quite possible to move, I'm not sure I like the idea. sqm-scripts is primarily targeted at LEDE/openwrt, so there's fairly tight coupling between the luci parts and the rest. Splitting that up would be a hassle, and I fear it would lead to breakage. Having both packages built from the same Makefile ensures that they are always in sync.

So could we make an exception here and keep luci-app-sqm in the packages repo? :)

danrl, 2017-02-15:
I experienced something similar with the wireguard packages. Yes, pushing to both repos is a bit annoying, but usually things are merged in a reasonable timeframe. It might soon really be the last package that is built from the "wrong" repository. Hmm...

tohojo, 2017-02-15:
And what, exactly, is the problem with having it in the "wrong" repo? :)

hnyman, 2017-02-15:
I don't see luci-app-sqm staying here attached to the main SQM Makefile as a big problem, as its LuCI source code is in the upstream repo. Only the Makefile is here in packages.

danrl, 2017-02-15:
> And what, exactly, is the problem with having it in the "wrong" repo? :)

For example: Changes that affect many or all packages need to be transferred over to this Makefile as well in such a case. Other than that, it's just for having a clean separation between the two projects.

tohojo, 2017-02-15:
Right, well in that case I'd prefer to keep the packages together. If a concrete problem arises from it we can reevaluate then. Good enough? :)

danrl, 2017-02-16:
At least I tried :)
It's a reasonable solution. Thanks for considering! I'll close this one, then.

closed 2017-02-16 by danrl