openwrt/packages #4020: python urrlib3 not found during LEDE build

noilly, 2017-02-16:
Maintainer: unsure (don't know which python package is failing)
Environment: LEDE
Description:

I am trying to build LEDE and include python in the build, but it fails, saying it can't find module `urllib3`. The error output is here: [http://pastebin.com/1QReeuPw](http://pastebin.com/1QReeuPw)

hnyman, 2017-02-16:
python stuff usually belongs to @commodo 

I just built python for ipq806x successfully (in Ubuntu x64 16.10) and also buildbots build practically all python modules, so that problem may be related to your buildhost & environment.

noilly, 2017-02-16:
What do you mean by buildhost?

This issue was opened in LEDE but they have [claimed](https://bugs.lede-project.org/index.php?do=details&task_id=511) this is because of a problem with the package, and an issue should be raised here. The original reporter of this problem is not myself, so I am not the only one experiencing this issue.

commodo, 2017-02-17:
@noilly 
Looks like an issue with your openssl devel libs on your host probably.
Ideally, the build should use the `libressl` package that comes with LEDE.

Can you tell me what distro are you using ?
Which LEDE commit you are on ?
Which packages repo commit you are on ?
Or, if you could provide me with the `python` make files [i.e. the `lang/python` ] ; I could try to use them.

This error was an issue about a month ago, but was fixed since then.

noilly, 2017-02-17:
@commodo 
I am building on Ubunutu 12.04
I'm on LEDE trunk: 2bf9ea6

Regarding packages repo commit,  how do I find out? I'm simply running:
```
./scripts/feeds update -a
./scripts/feeds install -a
```

**feeds/packages/lang/python/Makefile:**
```
#
# Copyright (C) 2006-2016 OpenWrt.org
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk

# For PYTHON_VERSION
include ./files/python-version.mk

PKG_NAME:=python
PKG_VERSION:=$(PYTHON_VERSION).$(PYTHON_VERSION_MICRO)
PKG_RELEASE:=4

PKG_SOURCE:=Python-$(PKG_VERSION).tar.xz
PKG_SOURCE_URL:=https://www.python.org/ftp/python/$(PKG_VERSION)
PKG_MD5SUM:=53b43534153bb2a0363f08bae8b9d990
PKG_HASH:=35d543986882f78261f97787fd3e06274bfa6df29fac9b4a94f73930ff98f731

PKG_LICENSE:=PSF
PKG_LICENSE_FILES:=LICENSE Modules/_ctypes/libffi_msvc/LICENSE Modules/_ctypes/darwin/LICENSE Modules/_ctypes/libffi/LICENSE Modules/_ctypes/libffi_osx/LICENSE Tools/pybench/LICENSE

# This file provides the necsessary host build variables
include ./files/python-host.mk

# For PyPackage
include ./files/python-package.mk

PKG_INSTALL:=1
PKG_BUILD_PARALLEL:=1
HOST_BUILD_PARALLEL:=1

PKG_BUILD_DIR:=$(BUILD_DIR)/Python-$(PKG_VERSION)
HOST_BUILD_DIR:=$(BUILD_DIR_HOST)/Python-$(PKG_VERSION)

PKG_BUILD_DEPENDS:=python/host
HOST_BUILD_DEPENDS:=bzip2/host expat/host

include $(INCLUDE_DIR)/host-build.mk
include $(INCLUDE_DIR)/package.mk

define Package/python/Default
  SUBMENU:=Python
  SECTION:=lang
  CATEGORY:=Languages
  TITLE:=Python $(PYTHON_VERSION) programming language
  URL:=https://www.python.org/
  MAINTAINER:=Alexandru Ardelean <ardeleanalex@gmail.com>
endef

define Package/python/Default/description
 Python is a dynamic object-oriented programming language that can be used
 for many kinds of software development. It offers strong support for
 integration with other languages and tools, comes with extensive standard
 libraries, and can be learned in a few days. Many Python programmers
 report substantial productivity gains and feel the language encourages
[... 228 lines cut ...]
        CONFIG_SITE=

define Host/Install
        $(MAKE) -C $(HOST_BUILD_DIR) install
        $(INSTALL_DIR) $(HOST_PYTHON_DIR)/bin/
        $(INSTALL_BIN) $(HOST_BUILD_DIR)/Parser/pgen $(HOST_PYTHON_DIR)/bin/pgen2
endef

$(eval $(call HostBuild))

$(foreach package, $(PYTHON_PACKAGES),  \
        $(eval $(call PyPackage,$(package))) \
        $(eval $(call BuildPackage,$(package))) \
)

$(eval $(call PyPackage,python-base))
$(eval $(call PyPackage,python-light))
$(eval $(call PyPackage,python))

$(eval $(call BuildPackage,python-pip-conf))

$(eval $(call BuildPackage,python-base))
$(eval $(call BuildPackage,python-light))
$(eval $(call BuildPackage,python))
```

commodo, 2017-02-17:
Curious;
what do you get for `aptitude show libssl-dev`
I have something like on XUbuntu 16.04
```
Package: libssl-dev                      
State: installed
Automatically installed: no
Multi-Arch: same
Version: 1.0.2g-1ubuntu4.6
Priority: optional
Section: libdevel
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Architecture: amd64
Uncompressed Size: 7.197 k
Depends: libssl1.0.0 (= 1.0.2g-1ubuntu4.6), zlib1g-dev
Recommends: libssl-doc
Breaks: libssl-dev:i386 (!= 1.0.2g-1ubuntu4.6)
Replaces: libssl-dev:i386 (< 1.0.2g-1ubuntu4.6)
Description: Secure Sockets Layer toolkit - development files
 This package is part of the OpenSSL project's implementation of the SSL and TLS cryptographic protocols for secure communication over the Internet. 
 
 It contains development libraries, header files, and manpages for libssl and libcrypto.
```

noilly, 2017-02-17:
`aptitude show libssl-dev`
```
Package: libssl-dev
State: installed
Automatically installed: no
Version: 1.0.1-4ubuntu5.39
Priority: optional
Section: libdevel
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Architecture: amd64
Uncompressed Size: 6,317 k
Depends: libssl1.0.0 (= 1.0.1-4ubuntu5.39), zlib1g-dev
Recommends: libssl-doc
Conflicts: libssl-dev
Description: SSL development libraries, header files and documentation
 libssl and libcrypto development libraries, header files and manpages.

 It is part of the OpenSSL implementation of SSL.
```

commodo, 2017-02-17:
@noilly 
back with some findings

the real error is a bit deeper than the above mentioned

first, the host Python does not build the `_hashlib.so` module
```
*** WARNING: renaming "_ssl" since importing it failed: build/lib.linux-x86_64-2.7/_ssl.so: undefined symbol: clock_gettime
building '_hashlib' extension
```

because of that, some stuff will cause failures

the reason `_hashlib.so` isn't built, is because of libressl ; which needs `clock_gettime()`
```
configure:13291: gcc -std=gnu99 -o conftest -O2 -I/home/sandu/lede/staging_dir/host/include -I/home/sandu/lede/staging_dir/host/usr/include  -fpic -Wall -std=gnu99 -fno-strict-aliasing  -fno-strict-overflow -D_FORTIFY_SOURCE=2 -fstack-protector-all -DHAVE_GNU_STACK  -Wno-pointer-sign -I/home/sandu/lede/staging_dir/host/include -I/home/sandu/lede/staging_dir/host/usr/include  -D_DEFAULT_SOURCE -D_BSD_SOURCE -D_POSIX_SOURCE -D_GNU_SOURCE -L/home/sandu/lede/staging_dir/host/lib -L/home/sandu/lede/staging_dir/host/usr/lib   -Wl,-z,relro -Wl,-z,now  conftest.c  >&5
/tmp/cckMN3ll.o: In function `main':
conftest.c:(.text.startup+0x15): undefined reference to `clock_gettime'
collect2: ld returned 1 exit status
configure:13291: $? = 1
```

i wonder if this can somehow be fixed via some LDFLAGS

commodo, 2017-02-17:
no luck yet ;
i tried
```
 -HOST_CONFIGURE_ARGS += --disable-shared
+HOST_CONFIGURE_ARGS += --disable-shared ac_cv_search_clock_gettime=no
 HOST_CFLAGS += $(FPIC)
```

but it didn't help yet ;
i think the fix may be to disable `clock_gettime()` in `libressl` or find some way to work around it

noilly, 2017-02-18:
@commodo thank you for looking into this! Is there a workaround I can use until then?

commodo, 2017-02-18:
you could try to revert this commit https://github.com/lede-project/source/commit/f6e6341d896adb78f9b496f71aab8f45e1742d5a
that should make use of Ubuntu's  `libopenssl-dev`  package

noilly, 2017-02-18:
@commodo  thank you! I always forget the command to that. What's the command do clean the directory?

commodo, 2017-02-18:
it's `make dirclean` 
not sure if it works to just `rm -rf staging_dir` ; you might get weird failures

noilly, 2017-02-18:
@commodo thank you! I usually remove the whole git repo and start again, which is a massive pain

commodo, 2017-02-20:
@noilly 
can you try this patch : https://github.com/commodo/packages/commit/5b7ace25aad01b7b960743b19be72e41fe758bda
it worked for me ; i tried also a clean build (please with a clean build ; if testing )

if you validate it, i will push it to Python(3) ; 

thanks
Alex

noilly, 2017-04-10:
@commodo sorry for the delay in reply, only now just getting back into the tech side of things.

Sorry for the naivety, how do I pull packages from a specific commit?

commodo, 2017-04-11:
well, you could 
* clone my repo and checkout the `python-link-issues-ubuntu-12-04` branch
* or if you feel you could do it manually from here https://github.com/commodo/packages/tree/python-link-issues-ubuntu-12-04/lang/python
that may also work
* but I think if you just try to add that line manually to your Makefile, it may also work

note that there have been a few updates to the Python packages [ since we last discussed ]
i am not sure any of them fixed this issue tho ;

commodo, 2017-05-26:
@noilly 
any more input on this ?

closed 2017-07-03 by jow-