openwrt/packages #4021: coreutils-base64 not available in menuconfig

noilly, 2017-02-16:
Maintainer: @jow- 
Environment: LEDE

The coreutils-base64 package is not available via menuconfig.

Installing coreutils through menuconfig does not install coreutils-base64.

Please make coreutils-base64 available via menuconfig.

hnyman, 2017-02-16:
coreutils-base64 is available in menuconfig.

First select coreutils top package, then go to the submenu and selectd coreutils-base64
```
  │ │                  < > cmdpad..... execute commands when a key is pressed, released or hold down           │ │  
  │ │                  -*- collectd................. Lightweight system statistics collection daemon  --->     │ │  
  │ │                  <M> coreutils......................................... The GNU core utilities  --->     │ │  
  │ │                  < > crelay................................... USB relay remote control daemon           │ │  
```

```
  │                                                                                                              │  
  │ ┌──────────────────────────────────────────────────────────────────────────────────────────────────────────┐ │  
  │ │                  --- coreutils......................................... The GNU core utilities           │ │  
  │ │                  <M>   coreutils-base64.............. Utility base64 from the GNU core utilities         │ │  
  │ │                  < >   coreutils-basename.......... Utility basename from the GNU core utilities (NEW)   │ │  
  │ │                  < >   coreutils-cat.................... Utility cat from the GNU core utilities (NEW)   │ │  
  │ │                  < >   coreutils-chcon................ Utility chcon from the GNU core utilities (NEW)   │ │  
  │ │                  < >   coreutils-chgrp................ Utility chgrp from the GNU core utilities (NEW)   │ │  

```

closed 2017-02-16 by hnyman

noilly, 2017-02-16:
Thank you! I did not see the sub-menu.