openwrt/packages #4033: Python3-ncurses: compilation failed

D-Albers, 2017-02-18:
Maintainer: @commodo 
Environment: LEDE 17.01 last git version

Description:

The compilation of the package failed with the following error message:

```
/opt/lede/17.01/build_dir/target-mips_24kc_musl-1.1.16/Python-3.6.0/Modules/_cursesmodule.c:119:12: error: conflicting types for 'setupterm'
 extern int setupterm(char *,int,int *);
            ^
In file included from ./Include/py_curses.h:53:0,
                 from /opt/lede/17.01/build_dir/target-mips_24kc_musl-1.1.16/Python-3.6.0/Modules/_cursesmodule.c:113:
/opt/lede/17.01/staging_dir/target-mips_24kc_musl-1.1.16/usr/include/term.h:740:28: note: previous declaration of 'setupterm' was here
 extern NCURSES_EXPORT(int) setupterm (NCURSES_CONST char *,int,int *);
                            ^

Python build finished successfully!
The necessary bits to build these optional modules were not found:
_tkinter              nis                                      
To find the necessary bits, look in setup.py in detect_modules() for the module's name.

The following modules found by detect_modules() in setup.py, have been
built by the Makefile instead, as configured by the Setup files:
atexit                pwd                   time               
zlib                                                           

Failed to build these modules:
_curses                                                        

Makefile:617: recipe for target 'sharedmods' failed
make[4]: *** [sharedmods] Error 1
```

commodo, 2017-02-18:
Looks familiar from Python.
Will check it out

commodo, 2017-02-20:
@D-Albers 
I fixed these issues in this PR https://github.com/openwrt/packages/pull/3998

I'll submit a PR for 17.01

commodo, 2017-02-20:
@D-Albers 
would it trouble you much if you can first check if the 2 patches [or just the Python3] patches work on your system ?

I've opened PR https://github.com/openwrt/packages/pull/4042

i'd prefer [ if possible ] to have a validation before merging

thanks
Alex

D-Albers, 2017-02-20:
@commodo 

I have seen your request to test the patches.
I will give them a run and will reply back at this thread.

D-Albers, 2017-02-20:
@commodo

The build has finished successful.

commodo, 2017-02-21:
@D-Albers @hnyman 
PR https://github.com/openwrt/packages/pull/4042 was merged
objections on closing this ?

thanks
Alex

closed 2017-02-21 by hnyman

shez1461, 2019-07-22:
Installing this version of ssl library helped solved the whole build issue I was facing.
possible that it could help anyone?
`sudo apt install -y libssl1.0-dev`

commodo, 2019-07-22:
> Installing this version of ssl library helped solved the exact build issue I was facing.
> possible that it could help anyone?
> `sudo apt install -y libssl1.0-dev`

well, that depends which version of OpenWrt you are using;
at some point in time, installing `libssl1.0-dev` on Debian/Ubuntu was required;
if it is required in the latest OpenWrt build, then it's probably a bug;