openwrt/packages #4048: ser2net: v3.3 - Segmentation fault (Lede on TL-WR1043v3)

walx666, 2017-02-21:
If you start ser2net with configline like :
`ser2net -C 4001:raw:300:/dev/ttyUSB0:57600:NONE:1STOPBIT:8DATABITS:-XONXOFF:-RTSCTS:LOCAL -u`

you will get “Segmentation fault”

Starting ser2net without argument seems to be ok.

On OpenWRT 15.05.1 and earlier, there was not a problem with that commandline.

`root@test-biom-r1:~#ser2net -C 4001:raw:300:/dev/ttyUSB0:57600:NONE:1STOPBIT:8DATABITS:-XONXOFF:-RTSCTS:LOCAL -u`
**`Segmentation fault`**
`root@test-lede:~# logread | tail`
`Mon Feb 20 14:01:04 2017 user.err : Unknown config item: 57600:NONE:1STOPBIT:8DATABITS:-XONXOFF:-RTSCTS:LOCAL on line 0`
`Mon Feb 20 14:01:04 2017 kern.info kernel: [ 4587.292580]`
`Mon Feb 20 14:01:04 2017 kern.info kernel: [ 4587.292580] do_page_fault(): sending SIGSEGV to ser2net for invalid read access from 000051c4`
`Mon Feb 20 14:01:04 2017 kern.info kernel: [ 4587.301306] epc = 00408f9d in ser2net[400000+f000]`
`Mon Feb 20 14:01:04 2017 kern.info kernel: [ 4587.306243] ra  = 0040954d in ser2net[400000+f000]`
`Mon Feb 20 14:01:04 2017 kern.info kernel: [ 4587.311178]`
`root@test-lede:~# ls /dev/ttyUSB*`
`/dev/ttyUSB0`

mhei, 2017-02-24:
I'm wondering whether this command line has valid syntax, because the part after the baudrate should be whitespace separated and not by colon... So this should work: `ser2net -C "4003:raw:300:/dev/ttyUSB0:115200 NONE 1STOPBIT 8DATABITS -XONXOFF -RTSCTS LOCAL" -u`
But I can reproduce the segmentation fault with ser2net v3.3 even on my local desktop system, but I cannot reproduce it with ser2net 3.4.
I recently updated the package feed to v3.4, could you test with this version again?

walx666, 2017-02-27:
Tested with ser2net v3.4 (lede-snapshot) > your configline seems to work on it.
I don't know, why i've used this configline (seperated by colon) - but it worked for me back then. Sadly the package repository for the first stable release of lede (LEDE 17.01 "Reboot"), has ser2net v3.3 included.
So i've changed my hotplug-script to use ser2net with a dynamically generated configfile.

mhei, 2017-03-16:
I'm closing this issue since it works with latest version. For first stable release, I'm still thinking about bumping ser2net to v3.4, but I'm not yet sure about it...

closed 2017-03-16 by mhei