openwrt/packages #4054: package/feeds/packages/hamlib: undefined reference to `usb_find_devices'

taeasy, 2017-02-22:
Maintainer: @acinonyx
Environment: LEDE v17.01.0

Description:

```
/home/main/lede/staging_dir/toolchain-x86_64_gcc-5.4.0_musl-1.1.16/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/../../../../x86_64-openwrt-linux-musl/bin/ar: `u' modifier ignored since `D' is the default (see `U')
make[5]: Leaving directory '/home/main/lede/build_dir/target-x86_64_musl-1.1.16/hamlib-3.0.1/yaesu'
Making all in amsat
make[5]: Entering directory '/home/main/lede/build_dir/target-x86_64_musl-1.1.16/hamlib-3.0.1/amsat'
  CC       if100.lo
  CCLD     libhamlib-amsat.la
/home/main/lede/staging_dir/toolchain-x86_64_gcc-5.4.0_musl-1.1.16/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/../../../../x86_64-openwrt-linux-musl/bin/ar: `u' modifier ignored since `D' is the default (see `U')
make[5]: Leaving directory '/home/main/lede/build_dir/target-x86_64_musl-1.1.16/hamlib-3.0.1/amsat'
Making all in ars
make[5]: Entering directory '/home/main/lede/build_dir/target-x86_64_musl-1.1.16/hamlib-3.0.1/ars'
  CC       ars.lo
  CCLD     libhamlib-ars.la
/home/main/lede/staging_dir/toolchain-x86_64_gcc-5.4.0_musl-1.1.16/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/../../../../x86_64-openwrt-linux-musl/bin/ar: `u' modifier ignored since `D' is the default (see `U')
make[5]: Leaving directory '/home/main/lede/build_dir/target-x86_64_musl-1.1.16/hamlib-3.0.1/ars'
Making all in celestron
make[5]: Entering directory '/home/main/lede/build_dir/target-x86_64_musl-1.1.16/hamlib-3.0.1/celestron'
  CC       celestron.lo
  CCLD     libhamlib-celestron.la
/home/main/lede/staging_dir/toolchain-x86_64_gcc-5.4.0_musl-1.1.16/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/../../../../x86_64-openwrt-linux-musl/bin/ar: `u' modifier ignored since `D' is the default (see `U')
make[5]: Leaving directory '/home/main/lede/build_dir/target-x86_64_musl-1.1.16/hamlib-3.0.1/celestron'
Making all in cnctrk
make[5]: Entering directory '/home/main/lede/build_dir/target-x86_64_musl-1.1.16/hamlib-3.0.1/cnctrk'
  CC       cnctrk.lo
  CCLD     libhamlib-cnctrk.la
/home/main/lede/staging_dir/toolchain-x86_64_gcc-5.4.0_musl-1.1.16/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/../../../../x86_64-openwrt-linux-musl/bin/ar: `u' modifier ignored since `D' is the default (see `U')
make[5]: Leaving directory '/home/main/lede/build_dir/target-x86_64_musl-1.1.16/hamlib-3.0.1/cnctrk'
Making all in easycomm
make[5]: Entering directory '/home/main/lede/build_dir/target-x86_64_musl-1.1.16/hamlib-3.0.1/easycomm'
  CC       easycomm.lo
  CCLD     libhamlib-easycomm.la
/home/main/lede/staging_dir/toolchain-x86_64_gcc-5.4.0_musl-1.1.16/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/../../../../x86_64-openwrt-linux-musl/bin/ar: `u' modifier ignored since `D' is the default (see `U')
make[5]: Leaving directory '/home/main/lede/build_dir/target-x86_64_musl-1.1.16/hamlib-3.0.1/easycomm'
Making all in ether6
make[5]: Entering directory '/home/main/lede/build_dir/target-x86_64_musl-1.1.16/hamlib-3.0.1/ether6'
  CC       ether6.lo
  CCLD     libhamlib-ether6.la
/home/main/lede/staging_dir/toolchain-x86_64_gcc-5.4.0_musl-1.1.16/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/../../../../x86_64-openwrt-linux-musl/bin/ar: `u' modifier ignored since `D' is the default (see `U')
make[5]: Leaving directory '/home/main/lede/build_dir/target-x86_64_musl-1.1.16/hamlib-3.0.1/ether6'
Making all in fodtrack
make[5]: Entering directory '/home/main/lede/build_dir/target-x86_64_musl-1.1.16/hamlib-3.0.1/fodtrack'
  CC       fodtrack.lo
  CCLD     libhamlib-fodtrack.la
/home/main/lede/staging_dir/toolchain-x86_64_gcc-5.4.0_musl-1.1.16/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/../../../../x86_64-openwrt-linux-musl/bin/ar: `u' modifier ignored since `D' is the default (see `U')
make[5]: Leaving directory '/home/main/lede/build_dir/target-x86_64_musl-1.1.16/hamlib-3.0.1/fodtrack'
Making all in gs232a
make[5]: Entering directory '/home/main/lede/build_dir/target-x86_64_musl-1.1.16/hamlib-3.0.1/gs232a'
  CC       gs232a.lo
  CC       gs232b.lo
  CC       gs232.lo
  CCLD     libhamlib-gs232a.la
/home/main/lede/staging_dir/toolchain-x86_64_gcc-5.4.0_musl-1.1.16/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/../../../../x86_64-openwrt-linux-musl/bin/ar: `u' modifier ignored since `D' is the default (see `U')
make[5]: Leaving directory '/home/main/lede/build_dir/target-x86_64_musl-1.1.16/hamlib-3.0.1/gs232a'
Making all in heathkit
make[5]: Entering directory '/home/main/lede/build_dir/target-x86_64_musl-1.1.16/hamlib-3.0.1/heathkit'
  CC       hd1780.lo
  CCLD     libhamlib-heathkit.la
/home/main/lede/staging_dir/toolchain-x86_64_gcc-5.4.0_musl-1.1.16/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/../../../../x86_64-openwrt-linux-musl/bin/ar: `u' modifier ignored since `D' is the default (see `U')
make[5]: Leaving directory '/home/main/lede/build_dir/target-x86_64_musl-1.1.16/hamlib-3.0.1/heathkit'
Making all in m2
make[5]: Entering directory '/home/main/lede/build_dir/target-x86_64_musl-1.1.16/hamlib-3.0.1/m2'
  CC       rc2800.lo
  CCLD     libhamlib-m2.la
/home/main/lede/staging_dir/toolchain-x86_64_gcc-5.4.0_musl-1.1.16/lib/gcc/x86_64-openwrt-linux-musl/5.4.0/../../../../x86_64-openwrt-linux-musl/bin/ar: `u' modifier ignored since `D' is the default (see `U')
make[5]: Leaving directory '/home/main/lede/build_dir/target-x86_64_musl-1.1.16/hamlib-3.0.1/m2'
[... 82 lines cut ...]
Makefile:564: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory '/home/main/lede/build_dir/target-x86_64_musl-1.1.16/hamlib-3.0.1'
Makefile:99: recipe for target '/home/main/lede/build_dir/target-x86_64_musl-1.1.16/hamlib-3.0.1/.built' failed
make[3]: *** [/home/main/lede/build_dir/target-x86_64_musl-1.1.16/hamlib-3.0.1/.built] Error 2
make[3]: Leaving directory '/home/main/lede/feeds/packages/utils/hamlib'
package/Makefile:105: recipe for target 'package/feeds/packages/hamlib/compile' failed
make[2]: *** [package/feeds/packages/hamlib/compile] Error 2
make[2]: Leaving directory '/home/main/lede'
package/Makefile:101: recipe for target '/home/main/lede/staging_dir/target-x86_64_musl-1.1.16/stamp/.package_compile' failed
make[1]: *** [/home/main/lede/staging_dir/target-x86_64_musl-1.1.16/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/main/lede'
/home/main/lede/include/toplevel.mk:197: recipe for target 'world' failed
make: *** [world] Error 2
main@debian:~/lede$ ./scripts/feeds update
Updating feed 'packages' from 'https://git.lede-project.org/feed/packages.git^ed90827282851ad93294e370860320f1af428bb2' ...
Create index file './feeds/packages.index' 
Updating feed 'luci' from 'https://git.lede-project.org/project/luci.git^a100738163585ae1edc24d832ca9bef1f34beef0' ...
Create index file './feeds/luci.index' 
Updating feed 'routing' from 'https://git.lede-project.org/feed/routing.git^dd36dd47bbd75defcb3c517cafe7a19ee425f0af' ...
Create index file './feeds/routing.index' 
Updating feed 'telephony' from 'https://git.lede-project.org/feed/telephony.git^1f0fb2538ba6fc306198fe2a9a4b976d63adb304' ...
Create index file './feeds/telephony.index' 
main@debian:~/lede$
```

Dinkar520, 2017-04-21:
i think need add libusb-compat.
[Makefile](https://github.com/openwrt/packages/files/946071/Makefile.txt)

wwtk, 2017-11-20:
that does fix it, not sure why it isnt in the trunk yet

dibdot, 2018-08-17:
Probably fixed nowadays.

closed 2018-08-17 by dibdot