openwrt/packages #4062: mwan3 is ignoring l2tp interface

franke2, 2017-02-23:
Maintainer: @Adze1502
Environment: D-Link DIR-860L B1 LEDE Reboot 17.01.0 r3205-59508e3 / LuCI lede-17.01 branch (git-17.051.53299-a100738)

Description:

mwan3 is not working with l2tp interface. 
it is not creating ip rule and routing table for that device.
interface is marked as down and is not able to ping.

my l2tp interface configuration is:

config interface 'wan4'
	option proto 'l2tp'
	option ifname 'eth0.4'
	option delegate '0'
	option server 'SERVER'
	option username 'USER'
	option password 'PASSWORD'
	option metric '40'
	option peerdns '0'
	option dns '8.8.8.8 8.8.4.4'
	option ipv6 '0'

feckert, 2017-03-18:
@franke2 could you please send the mwan3 config as well?

feckert, 2018-04-03:
@hnyman I think we could close this as well?

closed 2018-04-03 by hnyman