openwrt/packages #4071: unbound 1.6.0-3: Generates invalid config if no ULA-Prefix is set

RalfJung, 2017-02-25:
Maintainer: @EricLuehrsen
Environment: TP-Link Archer C7 v2, LEDE Reboot 17.01.0 r3205-59508e3 / LuCI lede-17.01 branch (git-17.051.53299-a100738)

Description:
The unbound server fails to start up because of an invalid configuration file. The file contains the following invalid line:
```
local-data: "nina. 120 IN AAAA 192.168.178.3"
```
This is bogus because `192.168.178.3` is clearly not an IPv6 address.

The reason this happens is the following check in `/usr/lib/unbound/unbound.sh`:
```
        "${ulaprefix%%:/*}"*)
          # Only this networks ULA and only hostname
          echo "  local-data: \"$UNBOUND_TXT_HOSTNAME. 120 IN AAAA $address\""
          echo "  local-data-ptr: \"$address 120 $UNBOUND_TXT_HOSTNAME\""
          ;;
```

We do not want ULAs to be distributed in the network (we already have publicly routable addresses, so we don't see the point). Hence we set the ULA in the interface configuration page to be empty. Unfortunately, that makes the check above behave incorrectly.

For now, we fixed the problem by commenting out the code block above. That's obviously not a long-term solution though.

EricLuehrsen, 2017-02-25:
Agree "is empty" or otherwise stuffed with something invalid from UCI should be a check here. I will look into it.

On a side note of caution, most organizations should have ULA configured unless one actually "owns" a static IP prefix segment and the number is in the ISP contract. The top-level-router needs some anchor for a prefix to delicate. Otherwise an org. is at the mercy of DHCP-PD from the ISP. If they are developing internal services or applications that are IP6 dependent (direct connectivity), then it could be ugly when the ISP has difficulty. I personally find it convenient to parallel the IP4 private network . `192.168.0.0=fd00:c0a8::` or `172.16.0.0=fd00:ac10::`

EricLuehrsen, 2017-02-25:
I agree "is empty" should be managed for that external UCI option. Try PR #4059.

On a side note of caution, most organizations should have ULA configured unless one actually "owns" a static IP prefix segment and the number is in the ISP contract. The top-level-router needs some anchor for a prefix to delicate. Otherwise an org. is at the mercy of DHCP-PD from the ISP. If they are developing internal services or applications that are IP6 dependent (direct connectivity), then it could be ugly when the ISP has difficulty. I personally find it convenient to parallel the IP4 private network . 192.168.0.0=fd00:c0a8:: or 172.16.0.0=fd00:ac10::

RalfJung, 2017-02-25:
> I agree "is empty" should be managed for that external UCI option. Try PR #4072.

How can I best try this PR? I could e.g. just copy the block that assigns `ulaprefix="fdno:such:addr::/48"` to the file on the device. (It'll be more than a week though until I have access to the device again.)

> The top-level-router needs some anchor for a prefix to delicate. Otherwise an org. is at the mercy of DHCP-PD from the ISP. If they are developing internal services or applications that are IP6 dependent (direct connectivity), then it could be ugly when the ISP has difficulty.

Ah, that makes sense. Finally I understand why people may want an ULA despite having public IPv6. Anyway, our network is rather small and whenever we absolutely had to configure a static address somewhere we just used link-local IPv6 or IPv4...

EricLuehrsen, 2017-02-25:
You said you commented out the lines of code, so I don't know if you did that live or one of the build processes. This one in particular is an easy live cut-n-paste.

closed 2017-02-26 by hnyman

RalfJung, 2017-02-27:
> You said you commented out the lines of code

Right, I did this live. But your PR also makes other changes to this file, and changes other files, so Iw wasn't sure what to use of that patch.

PR #4059 (2017-02-26): unbound: DNSSEC trust anchor roll for 2017
Maintainer: me
Compile Tested: LEDE Trunk
Run Tested: TL-Archer-C7

Carry both the 2010 and 2017 DS records until 2018.
Unbound complies with RFC5011 but fresh installs
need these anchors to get started.
https://www.icann.org/resources/pages/ksk-rollover
https://www.iana.org/domains/root

Bug fix: #4071