openwrt/packages #4076: Override mac address doesn't work on wan

heiybb, 2017-02-27:
Router:WNDR4300 v1
LEDE Version:lede-17.01.0-r3205-59508e3

As what the title describes, and there is nothing wrong in the config file.

sholwe, 2017-03-18:
Confirmed.

Just did a clean sysupgrade install from factory, and when I initially set the MAC override, it displayed it.  It continues to show in LuCI after a reboot, but does not actually reflect it anywhere.

fehead, 2017-05-09:
Same probleam is happend.

root@LEDE:/etc/config# cat /etc/openwrt_release 
DISTRIB_ID='LEDE'
DISTRIB_RELEASE='17.01.1'
DISTRIB_REVISION='r3316-7eb58cf109'
DISTRIB_CODENAME='reboot'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='LEDE Reboot 17.01.1 r3316-7eb58cf109'
DISTRIB_TAINTS='no-all'

So I edit /etc/config/network file.

config device 'wan_dev'
        option name 'eth0.2'
        option macaddr 'XX:XX:XX:XX:XX:XX'   # <-- Add 

and reboot. and It works well.

closed 2018-08-17 by dibdot