openwrt/packages #4085: net-snmp: Request to compile with support for the "extend" module.

sorano, 2017-02-28:
Maintainer: @stintel  (find it by checking history of the package Makefile)
Environment: (mips)

Description:

Would be nice if net-snmp was compiled with support for the extend module so external scripts can be called to be used with monitoring tools.

--with-mib-modules="agent/extend" should be added to the CONFIGURE_OPTS variable in the net-snmp makefile.

http://www.net-snmp.org/docs/man/snmpd.conf.html#lbAZ

stintel, 2017-02-28:
Please test https://github.com/stintel/openwrt-packages/commit/565b070c8b0856f757ca62ccf687b7af8e80e8f0

sorano, 2017-03-01:
I've tested now and it works.

Good job 👍

closed 2017-03-01 by stintel

davidrs16, 2017-11-06:
Hi,

I need the snmp extend functionality into an arduino yun that use openwrt as OS.
What should I do to install your modified version of net-snmp package into the arduino yun?

Should I add this software respository and install with opkg manager? Or may be should I compile it? If so please could someone give me some help with the basic steps to compile and install the package?

Thank you in advance!

sorano, 2017-11-06:
This has been merged in standard OpenWrt packages feed.

So as long as your OS uses that feed you should be fine.

```
opkg list snmpd
snmpd - 5.7.3-7
```

davidrs16, 2017-11-06:
Thank you sorano for your repond.
I´m using arduino yun that used Openwrt as OS, but when I tried to **opkg list snmpd** I get the snmpd version 5.4.2.1-5
Should I change somthing in the opkg repositories to force opkg to fin the 5.7 version?

Thank you very much!

sorano, 2017-11-06:
Probably Arduino Yun uses another feed for the packages? Or you have not updated the list?
Check which url you use for your list. Then browse that list manually to see version there.

davidrs16, 2017-11-06:
Hi again Sorano,

Yes you are right the arduino opkg.conf looks like, and snmpd version of this respository is 5.4.2.1:

**src/gz attitude_adjustment http://downloads.arduino.cc/openwrtyun/1/packages
dest root /
dest ram /tmp
lists_dir ext /var/opkg-lists
option overlay_root /overlay
option check_signature**

I have tried to add this repositories and disable signature check:

**src/gz attitude_adjustment http://downloads.arduino.cc/openwrtyun/1/packages
dest root /
dest ram /tmp
lists_dir ext /var/opkg-lists
option overlay_root /overlay
src/gz chaos_calmer_base http://downloads.openwrt.org/snapshots/trunk/ar71xx/gen
src/gz chaos_calmer_packages http://downloads.openwrt.org/snapshots/trunk/ar71xx
src/gz chaos_calmer_luci http://downloads.openwrt.org/snapshots/trunk/ar71xx/gen
src/gz chaos_calmer_routing http://downloads.openwrt.org/snapshots/trunk/ar71xx/
src/gz chaos_calmer_telephony http://downloads.openwrt.org/snapshots/trunk/ar71x
src/gz chaos_calmer_management http://downloads.openwrt.org/snapshots/trunk/ar71**

Doing this and executing opkg update I can see snmpd 5.7.3, but when I try to install I get the following error:

**# opkg install snmpd
Package snmpd (5.7.3-9) installed in root is up to date.
Configuring libnetsnmp.
//usr/lib/opkg/info/libnetsnmp.postinst: line 4: default_postinst: not found
Configuring snmp-utils.
//usr/lib/opkg/info/snmp-utils.postinst: line 4: default_postinst: not found
Configuring snmpd.
//usr/lib/opkg/info/snmpd.postinst: line 4: default_postinst: not found
Collected errors:
  pkg_run_script: package "libnetsnmp" postinst script returned status 127.
  opkg_configure: libnetsnmp.postinst returned 127.
  pkg_run_script: package "snmp-utils" postinst script returned status 127.
  opkg_configure: snmp-utils.postinst returned 127.
  pkg_run_script: package "snmpd" postinst script returned status 127.
  opkg_configure: snmpd.postinst returned 127.**

Do you know what can I do to install the 5.7.3 version?

Thank you very much!!!

karlp, 2017-11-06:
You cannot just try and run CC based images on your AA based image.  Complain to arduino/ yun that their "lininio" "openwrt" is abyssmally outdated.  It's a fork from years back of openwrt, and there's very little anyone here can do to provide support for them

karlp, 2017-11-06:
if you want to run _actual_ lede/openwrt on your yun, you can install http://downloads.lede-project.org/snapshots/targets/ar71xx/generic/lede-ar71xx-generic-arduino-yun-squashfs-sysupgrade.bin but that doesn't have any oof the lininio extras.

davidrs16, 2017-11-06:
Thanks Karlp,
I dont know nothing about lede project. I am going to take a look to the lede doc.
Do you know which linino extras are not availble? At least I need to talk with the mikrokontroller processor using the brigde API.
And is not possible to compile the snmpd package for the linino?

Thanks for your help, I am a little stuck with this monitoring project...your help is really appreciated.

commit ec6712fe (2017-03-01): net-snmp: enable agent/extend MIB

Closes #4085.