openwrt/packages #4087: net/ ntpd: ntpd.hotplug script makes disabling service impossible [CC]

the-real-tokai, 2017-02-28:
Maintainer: Peter Wagner <tripolar@gmx.at>
Environment: CC

Description: The provided `ntpd.hotplug` script (that is installed to `/etc/hotplug.d/iface/20-ntpd`) makes disabling the service impossible.

Basically what happens is that when the `ntpd` service gets disabled —and should stay disabled— at some point (e.g. during reboot or a reconnect) the hotplug script gets launched and fires `ntpd` up again. Not what we want in this case. 😏

Solution: Probably an extra check in the hotplug script for the respective ntpd file in `/etc/rc.d` could help?

EricLuehrsen, 2017-03-06:
Both dnsmasq and Unbound UCI check their respective enabled status. They use NTP hotplug to know when full DNSSEC is possible (requires global time). The following is just a simplified example.
```
if [ -x /etc/init.d/[app] ] ; then
  /etc/init.d/[app] enabled && /etc/init.d/[app] restart
fi
```

tripolar, 2017-03-06:
fixed in https://github.com/openwrt/packages/commit/228764bfe8b439e6abbd54eec3026a038e3c7a9c

closed 2017-03-06 by tripolar