openwrt/packages #4089: [REGRESSION] boost: 21091fa0acda ("boost: 1.63 Revision 3 -> Python 3.6 Support Fixed and Unsupported Target Fixes") breaks dependency check

seragh, 2017-03-01:
Hi Carlos,

Commit 21091fa0acda ("boost: 1.63 Revision 3 -> Python 3.6 Support Fixed
and Unsupported Target Fixes") breaks dependency check on openwrt/master
for boost-filesystem as follows:

```
Package boost-filesystem is missing dependencies for the following libraries:
libboost_system.so.1.63.0
```

```
$ grep boost .config
CONFIG_PACKAGE_boost=y
# CONFIG_boost-shared-libs is not set
# CONFIG_boost-static-libs is not set
CONFIG_boost-static-and-shared-libs=y
CONFIG_boost-runtime-shared=y
CONFIG_boost-variant-release=y
# CONFIG_boost-variant-debug is not set
# CONFIG_boost-variant-profile is not set
# CONFIG_boost-use-name-tags is not set
# CONFIG_boost-libs-all is not set
# CONFIG_boost-test-pkg is not set
# CONFIG_boost-coroutine2 is not set
# CONFIG_boost-graph-parallel is not set
# CONFIG_PACKAGE_boost-atomic is not set
# CONFIG_PACKAGE_boost-chrono is not set
# CONFIG_PACKAGE_boost-container is not set
# CONFIG_PACKAGE_boost-context is not set
# CONFIG_PACKAGE_boost-coroutine is not set
# CONFIG_PACKAGE_boost-date_time is not set
# CONFIG_PACKAGE_boost-fiber is not set
CONFIG_PACKAGE_boost-filesystem=y
# CONFIG_PACKAGE_boost-graph is not set
# CONFIG_PACKAGE_boost-iostreams is not set
# CONFIG_PACKAGE_boost-log is not set
# CONFIG_PACKAGE_boost-math is not set
CONFIG_PACKAGE_boost-program_options=y
# CONFIG_PACKAGE_boost-random is not set
# CONFIG_PACKAGE_boost-regex is not set
# CONFIG_PACKAGE_boost-serialization is not set
# CONFIG_PACKAGE_boost-signals is not set
CONFIG_PACKAGE_boost-system=y
# CONFIG_PACKAGE_boost-thread is not set
# CONFIG_PACKAGE_boost-timer is not set
# CONFIG_PACKAGE_boost-wave is not set
```

Reverting the mentioned commit fixes the issue.

Ralph

PS: The ("## If you are thinking in making changes to the following list, for
some reason, the order of what follows matters.") looks really fishy.

hnyman, 2017-03-01:
cc to package maintainer: @ClaymorePT

ClaymorePT, 2017-03-01:
Ok, thank you for the warning. 
That seems to be a minor issue with a dependency that I missed. I will work on that.

ClaymorePT, 2017-03-01:
Ok I have opened a push request to revert that commit -> https://github.com/openwrt/packages/pull/4091
Later, I will look into a different way to solve my primary issue.
Once again, thank you.

seragh, 2017-03-01:
Hi Carlos,

thanks for the fast "fix". Closing here.

Ralph

closed 2017-03-01 by seragh

ClaymorePT, 2017-03-01:
No problem. It was my fault. I forgot to fully  test it with OpenWRT. 
I will be more carefull next time.