openwrt/packages #4090: ulogd: Illegal instruction

LEDEuser, 2017-03-01:
Environment:
LEDE 17.01.0
LEDE_BOARD="ipq806x/generic"
LEDE_ARCH="arm_cortex-a15_neon-vfpv4"
ROUTER="Netgear R7800"

ulogd won't start.
When executing "/usr/sbin/ulogd" it returns "Illegal instruction"

Should I do extra configuration or is this to be fixed in a next build/release?

commodo, 2017-03-16:
well, there's a bug with ulogd that i've found
not sure if that's the case for you too

since we switched to musl & GCC 5.4 [ not sure which of these 2 causes the issue], ulogd was constantly crashing on start.

looking at the code, it seems there's a bug in ulogd with strings, where some memcpy() is being done, but there's no extra room for null char

will send a patch now

commodo, 2017-03-16:
@LEDEuser 
can you check this patch https://github.com/commodo/packages/commit/aca1f8bdd2117724a4651f282219a0b669c866ee 

it may fix the issue in your case
i'd say that the issue is related to https://github.com/openwrt/packages/issues/4123

LEDEuser, 2017-03-18:
Then first I have to figure out how to apply the patch and compile the package.
After that I will check the patch. Up til now I have only used the images.
Please have a little patience.

commodo, 2017-03-18:
I coud try to compile an image if you want.

I don't have your board/arch around.

LEDEuser, 2017-03-18:
Tomorrow I will look in to it and try to patch.
I have a Debian PC ready and it can't hurt to know how to compile.

commodo, 2017-03-18:
ok ; sure :)

LEDEuser, 2017-03-19:
Compiled a new .ipk file with the ipq806x SDK with your patch added.
Installed it with the command:
opkg install --force-reinstall --force-overwrite --force-checksum ulogd_2.0.5-1_arm_cortex-a15_neon-vfpv4.ipk

After that ulogd started and filled the log file. Only tested with a log file.

commodo, 2017-03-20:
ok ; then let's make an official PR for this

LEDEuser, 2017-03-20:
In the PR you write it fixes #4123. I commented that issue. Unfortunately it isn't fixed with this patch.

commodo, 2017-03-20:
My bad.

Will update

closed 2017-03-29 by thess