openwrt/packages #4114: Python 2: Samba 4.x compatibility

diizzyy, 2017-03-06:
@commodo 
Hi, is it possible to add these fixes (last on that page) to the repo? It makes Samba 4.x (not in tree atm) compile without replacing the current version.
https://github.com/wongsyrone/packages-1/commits/master/lang/python?author=wongsyrone

commodo, 2017-03-06:
@diizzyy 
short answer : yes
long answer, i'd have to walk through that patch

though, maybe it would be an idea, if you could provide me with a Samba Makefile, and I can validate the patches that I'll port ;
I'll give credit to @wongsyrone  for the patches/changes ; 
but it is a bit hard to walk through it, and harder to understand which one fixes the Samba build ;

wongsyrone, 2017-03-06:
The most related patch is:

010-fix-python-config.patch
https://github.com/wongsyrone/packages-1/commit/60ca6f0f2b6712e023bc4af38f15a10d1aa78154

commodo, 2017-03-06:
yeah, but that's still a pretty big patch ; 
it has whitespace in there

commodo, 2017-03-06:
i mean, i can just take in the parts that are not whitespace, but I'd like to see for myself that it fixes Samba

wongsyrone, 2017-03-06:
You can take https://raw.githubusercontent.com/wongsyrone/packages-1/master/lang/python/patches/010-fix-python-config.patch if you like.

For long story, you can check https://github.com/wongsyrone/openwrt-1/issues/73

wongsyrone, 2017-03-06:
The main point of this change is to ship a modified cross-compiling-friendly python2-config, and make use of that file to satisfy samba's waf, which is only compatible with python2.

https://github.com/wongsyrone/openwrt-1/blob/master/package/external/samba44related/samba44/Makefile#L69

commodo, 2017-03-06:
whoa, now i have my work cut out for myself 
will look into it

diizzyy, 2017-03-06:
Samba and Heimdal can be found here:
https://github.com/wongsyrone/openwrt-1/tree/master/package/external/samba44related/samba44
You need to add the following patches to Heimdal otherwise it'll fail since EGD isn't in OpenSSL.
https://github.com/wongsyrone/lede-1/issues/32#issuecomment-280840240

commodo, 2017-03-11:
@wongsyrone 
if you change
```
PYTHON_CONFIG="$(STAGING_DIR)/usr/bin/python2-config" \
```
to 
```
PYTHON_CONFIG="$(STAGING_DIR_HOSTPKG)/bin/python2-config" \
```

It builds on my machine.
Did you try it ?

I am not sure if it works though.
I tested on a x86_64 VM and Samba did not start; it said "Samba4 is disabled" ; maybe I need a configuration ?

Can you test ?
The only change I had to do to the Samba4 & Heimdal packages was that line.
Everything else is from here:
https://github.com/wongsyrone/openwrt-1/tree/master/package/external/samba44related/samba44

commodo, 2017-03-11:
@diizzyy maybe you could also try real quick :) ?

wongsyrone, 2017-03-11:
As long as I know, STAGING_DIR_HOSTPKG doesn't exist when I was testing samba4.

https://github.com/wongsyrone/openwrt-1/blob/master/package/external/samba44related/samba44/files/samba44.config#L2

You may change 'enabled' to 1 and try again. I add it to bypass autostart. https://github.com/wongsyrone/openwrt-1/blob/master/package/external/samba44related/samba44/files/samba44.init#L140

wongsyrone, 2017-03-11:
BTW, if you leave renice disabled in busybox configs, please remove all renice calls in the init script.

diizzyy, 2017-03-12:
@commodo 
Hmm... Samba isn't happy on ARM (Marvell)
```
Checking for program python2.7-config                                             : /home/diizzy/marvell-take2/source/staging_dir/hostpkg/bin/python2-config
Checking for custom code                                                          : Could not find the python development headers
```

commodo, 2017-03-22:
hello again
sorry for the delays ; 
seems stuff was pouring on all sides for me

i neglected to say earlier that I did reproduce the issue

please try this patch:
https://github.com/commodo/packages/commit/0bb78f682519efafeb69cae251bf1d158effbd51

i've managed to build on ARM ; i did not run it ;
can you try the patch on your setups ? 

thanks
Alex

diizzyy, 2017-03-22:
@commodo 
**Awesome work!** Many thanks! 👍 

[image]

commodo, 2017-03-22:
Cool :)
Glad it worked.

I'm also happy it worked with minimal changes.

<https://cloud.githubusercontent.com/assets/11659781/24220603/25f852f6-0f4b-11e7-82ba-f033fd2cd04a.png>

commodo, 2017-05-26:
@diizzyy 
can we close this issue ?
or is there anything else ?

diizzyy, 2017-05-26:
We can close this, thanks!

closed 2017-05-26 by diizzyy