openwrt/packages #4117: Report bug: transmission-web can't work in Safari mobile

JeremyPeng, 2017-03-07:
Package: <transmission-web 2.92+git-4 x86-64>

expected behavior:
It should be able to work properly in iOS10’s safari and chrome 56

actual behavior:

Only appear a web UI, without any seed-related lists and content, in iPhone7 iOS10 Safari
The seed list appears repeatedly and will gradually increase, in Windows 10 Chrome 56
Repair method:
I downlad transmission 2.92(14714)'s web folder and replaced it.
Everything is back to normal.
Hope to be able to re-release the package to fix this problem.

Platform:

QNAP TS-453A QTS 4.2.3

SimpleBobster, 2018-02-17:
@neheb Is the maintainer for transmission, but not sure about transmission-web. Seems like a better fix for transmission-web would be to build in into LuCI

neheb, 2018-02-17:
You have an old git version. In git this was fixed by: https://github.com/transmission/transmission/commit/bcfe147e6a41ce7774f569ede6179a7ed2c7f560#diff-90cb18eb85291b642ba26b968b3e210e

So basically log in to your router through ssh and edit that file.

Alternatively for latest stable, this patch should accomplish the same thing:

```
diff --git a/web/javascript/common.js b/web/javascript/common.js
index 7162d3f82..77bf96857 100644
--- a/web/javascript/common.js
+++ b/web/javascript/common.js
@@ -61,7 +61,7 @@ $(document).ready(function() {
                        // Fix min height for isMobileDevice when run in full screen mode from home screen
                        // so the footer appears in the right place
                        $('body div#torrent_container').css('min-height', '338px');
-               $("label[for=torrent_upload_url]").text("URL: ");
+               $("label[for=torrent_upload_url]").text = "URL: ";
        } else {
                // Fix for non-Safari-3 browsers: dark borders to replace shadows.
                $('div.dialog_container div.dialog_window').css('border', '1px solid #777');
```

MildBub, 2018-08-12:
I can confirm this issue is fixed in the latest 18.06 build and can be closed!

closed 2018-08-17 by dibdot