openwrt/packages #4123: ulogd: Trace/breakpoint trap

LEDEuser, 2017-03-08:
Environment:
LEDE 17.01.0
LEDE_BOARD="ar71xx/generic"
LEDE_ARCH="mips_24kc"
ROUTER="Netgear WNDR3800"

ulogd won't start.
When executing "/usr/sbin/ulogd" it returns "Trace/breakpoint trap"

Should I do extra configuration or is this to be fixed in a next build/release?

LEDEuser, 2017-03-19:
Compiled a new .ipk file with the ar71xx SDK with the patch from #4090.
Installed it with the command:
opkg install --force-reinstall --force-overwrite --force-checksum ulogd_2.0.5-1_mips_24kc.ipk

This resulted in the following message:
No packages removed.
Installing ulogd (2.0.5-1) to root...
Ignored ulogd sha256sum mismatch.
Segmentation fault

/usr/sbin/ulogd still gives
Trace/breakpoint trap

diizzyy, 2019-03-28:
Wrong repo, it's in OpenWrt's main repo.
Use https://bugs.openwrt.org/ instead

closed 2019-03-28 by diizzyy