openwrt/packages #4131: luci-app-mwan3: bad formatting of troubleshooting page

psyborg55, 2017-03-10:
Maintainer: @arfett 

Description:

when there is more troubleshooting data output in one line than fits into screen the page will look strange

[image]

i think it needs reformatting in syslog/kernel log page style

feckert, 2017-04-18:
That has to-do with the long output of `iptables`. The other views are effect as well if the screen is small enough
I am not sure how to fix this.
It is `css` magic.
I am not a `css` guru.
If you have a solution then go for it.

arfett, 2017-04-20:
I spent a chunk of time trying to fix that once. As you can see I was unable to. Maybe a css wizard can be found to work it out.

psyborg55, 2017-09-13:
not fixed with https://github.com/openwrt/luci/pull/1178/commits/e61f40ead8072fade1e2b0d24bf1ab8a195a22c6

it also happens on Configuration -> Interfaces, Rules tabs

feckert, 2017-11-03:
@psyborg55 i could nor reproduce your issue?
Could you send me a picture who does it looks?

feckert, 2018-03-15:
@psyborg55 have you tested the new luci master i have done some additional changes. 
If every thing is fine could you please close this issue?

feckert, 2018-04-03:
@hnyman i think we could close this issue there was a lot work done in the luci-app-mwan3 and the reported did not react on my ping

closed 2018-04-03 by hnyman