openwrt/packages #4149: transmissiondcfp: package (NEW)

jf-stretch, 2017-03-13:
please consideration _transmissiondcfp_
it's good stable
Optware-ng is not to sustain https://github.com/Optware/Optware-ng/issues/186
but now also not of  maintain _transmissiondcfp_   began stopped merge https://github.com/Optware/Optware-ng/pull/187/commits/d34985dfe21b45c6a33eac5d4ca00cb069e228e0

diizzyy, 2017-03-26:
If you need it please maintain your own package. Upstream works fine.

jf-stretch, 2017-03-27:
Ok, I not the ability of sustain maintain, but entware-ng for me the mutate. Original transmission daemon stalled at source one year insignificantly unchanged. transmissiondcfp remains stable. Thank you.

diizzyy, 2017-03-27:
No, it did not.
https://github.com/transmission/transmission/commits/master

jf-stretch, 2017-04-14:
> No, it did not.
> https://github.com/transmission/transmission/commits/master

Daemon of it withholds lacking enhancement features, that said remains stalled. Transmissiondcfp many of enhancement features new on daemon. It's good, each for his own.

diizzyy, 2017-06-02:
@jf-stretch 
Please close this request, if you want a new port this isn't the right forum.

closed 2017-06-02 by thess