openwrt/packages #4150: rt2x00: bug with mt7620 target power parsing from eeprom

psyborg55, 2017-03-14:
I’ve noticed with default auto txpower config 5GHz radio will get proper target power and use it. 2.4GHz radio just gets 20dBm as default from somewhere, which clearly isn’t eeprom since it has also target power 17dBm.

-Xiaomi Miwifi Mini
-LEDE Reboot SNAPSHOT r3742-894ee95
-OpenWrt R50104

dangowrt, 2017-03-14:
1. wrong place to file the bug. it's not a bug of a package, but of OpenWrt or LEDE or both
2. where do you read (or measure?) this value. default transmit power comes from regdb + eeprom and what is displayed in the webui or cli is EIRP **including** the antenna gain, **not** the raw value being set in TXALC. To see what gets programmed to TXALC, you can dump the register in debugfs (left as an excercise to the reader)

closed 2017-03-14 by dangowrt

psyborg55, 2017-03-15:
maybe there could be a way to move those wrong opened issues to relevant trees.

how could there be eirp displayed if the antenna gain is missing. mt76 displays exact target power from eeprom.

894ee95 checked out 2 days ago and built

dangowrt, 2017-03-15:
I found commit 894ee95 ipq806x: fix ethernet DMA parameters with linux 4.9
Sorry for the misunderstanding. Now about the issue you describe (let me phrase it as "_TX-power displayed for rt2x00 is always 100% of the allowed value in a reg-domain, independently of what's set in the EEPROM_"):
rt2x00 makes the assumption that device is calibrated to 100% tx power at the currently set regdomain. That screws up values being displayed. And mt76 does that differently. If you want to discuss this, discuss it on [the linux-wireless mailing list](https://wireless.wiki.kernel.org/en/developers/MailingLists) with all rt2x00 developers. It **is** an intersting discussion, because the Linux regulatory model and Ralink's approach are quite incompatible, there is no good solution which will guarantee keeping within power limits as well as displaying meaningful numbers to the user. They hence decided to only limit power by the offset and assume that the device is calibrated to power limits of the regulatory domain you set, which makes it behave exactly like the vendor driver but obviously makes it show wrong values for devices calibrated at less than 100% of the allowed limit or when setting a regulatory domain which is not the one you bought the device in. 
See [this comment in the code](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/net/wireless/ralink/rt2x00/rt2800lib.c#n3661).