openwrt/packages #4155: protobuf-c compiler error

ppaulojr, 2017-03-16:
Maintainer: @jsiverskog  @jow- 
Environment: ramips, mt7620, current commit

Description:
Error compiling protobuf-c

```
azureuser@compilemachine:~/build/openwrt-github/openwrt$ make package/protobuf-c/compile V=s
make[1]: Entering directory `/home/azureuser/build/openwrt-github/openwrt'
make[2]: Entering directory `/home/azureuser/build/openwrt-github/openwrt/package/libs/toolchain'
if [ -f /home/azureuser/build/openwrt-github/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.16/pkginfo/toolchain.default.install.clean ]; then rm -f /home/azureuser/build/openwrt-github/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.16/pkginfo/toolchain.default.install /home/azureuser/build/openwrt-github/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.16/pkginfo/toolchain.default.install.clean; fi; echo "libc" >> /home/azureuser/build/openwrt-github/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.16/pkginfo/toolchain.default.install
if [ -f /home/azureuser/build/openwrt-github/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.16/pkginfo/toolchain.default.install.clean ]; then rm -f /home/azureuser/build/openwrt-github/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.16/pkginfo/toolchain.default.install /home/azureuser/build/openwrt-github/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.16/pkginfo/toolchain.default.install.clean; fi; echo "libgcc" >> /home/azureuser/build/openwrt-github/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.16/pkginfo/toolchain.default.install
if [ -f /home/azureuser/build/openwrt-github/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.16/pkginfo/toolchain.default.install.clean ]; then rm -f /home/azureuser/build/openwrt-github/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.16/pkginfo/toolchain.default.install /home/azureuser/build/openwrt-github/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.16/pkginfo/toolchain.default.install.clean; fi; echo "libstdcpp" >> /home/azureuser/build/openwrt-github/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.16/pkginfo/toolchain.default.install
if [ -f /home/azureuser/build/openwrt-github/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.16/pkginfo/toolchain.default.install.clean ]; then rm -f /home/azureuser/build/openwrt-github/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.16/pkginfo/toolchain.default.install /home/azureuser/build/openwrt-github/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.16/pkginfo/toolchain.default.install.clean; fi; echo "libpthread" >> /home/azureuser/build/openwrt-github/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.16/pkginfo/toolchain.default.install
if [ -f /home/azureuser/build/openwrt-github/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.16/pkginfo/toolchain.default.install.clean ]; then rm -f /home/azureuser/build/openwrt-github/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.16/pkginfo/toolchain.default.install /home/azureuser/build/openwrt-github/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.16/pkginfo/toolchain.default.install.clean; fi; echo "librt" >> /home/azureuser/build/openwrt-github/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.16/pkginfo/toolchain.default.install
make[2]: Leaving directory `/home/azureuser/build/openwrt-github/openwrt/package/libs/toolchain'
make[2]: Entering directory `/home/azureuser/build/openwrt-github/openwrt/feeds/packages/libs/protobuf-c'
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kec -mdsp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /home/azureuser/build/openwrt-github/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.16/libprotobuf-c-v1.2.1:libprotobuf-c-v1.2.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/azureuser/build/openwrt-github/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.16/usr/include -I/home/azureuser/build/openwrt-github/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.16/include -I/home/azureuser/build/openwrt-github/openwrt/staging_dir/toolchain-mipsel_24kec+dsp_gcc-5.3.0_musl-1.1.16/usr/include -I/home/azureuser/build/openwrt-github/openwrt/staging_dir/toolchain-mipsel_24kec+dsp_gcc-5.3.0_musl-1.1.16/include/fortify -I/home/azureuser/build/openwrt-github/openwrt/staging_dir/toolchain-mipsel_24kec+dsp_gcc-5.3.0_musl-1.1.16/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kec -mdsp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /home/azureuser/build/openwrt-github/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.16/libprotobuf-c-v1.2.1:libprotobuf-c-v1.2.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/azureuser/build/openwrt-github/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.16/usr/include -I/home/azureuser/build/openwrt-github/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.16/include -I/home/azureuser/build/openwrt-github/openwrt/staging_dir/toolchain-mipsel_24kec+dsp_gcc-5.3.0_musl-1.1.16/usr/include -I/home/azureuser/build/openwrt-github/openwrt/staging_dir/toolchain-mipsel_24kec+dsp_gcc-5.3.0_musl-1.1.16/include/fortify -I/home/azureuser/build/openwrt-github/openwrt/staging_dir/toolchain-mipsel_24kec+dsp_gcc-5.3.0_musl-1.1.16/include " LDFLAGS="-L/home/azureuser/build/openwrt-github/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.16/usr/lib -L/home/azureuser/build/openwrt-github/openwrt/staging_dir/target-mipsel_24kec+dsp_musl-1.1.16/lib -L/home/azureuser/build/openwrt-github/openwrt/staging_dir/toolchain-mipsel_24kec+dsp_gcc-5.3.0_musl-1.1.16/usr/lib -L/home/azureuser/build/openwrt-github/openwrt/staging_dir/toolchain-mipsel_24kec+dsp_gcc-5.3.0_musl-1.1.16/lib -znow -zrelro " make -j1 -C /home/azureuser/build/openwrt-github/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.16/libprotobuf-c-v1.2.1/. AR="mipsel-openwrt-linux-musl-gcc-ar" AS="mipsel-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kec -mdsp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap /home/azureuser/build/openwrt-github/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.16/libprotobuf-c-v1.2.1:libprotobuf-c-v1.2.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mipsel-openwrt-linux-musl-ld NM="mipsel-openwrt-linux-musl-gcc-nm" CC="mipsel-openwrt-linux-musl-gcc" GCC="mipsel-openwrt-linux-musl-gcc" CXX="mipsel-openwrt-linux-musl-g++" RANLIB="mipsel-openwrt-linux-musl-gcc-ranlib" STRIP=mipsel-openwrt-linux-musl-strip OBJCOPY=mipsel-openwrt-linux-musl-objcopy OBJDUMP=mipsel-openwrt-linux-musl-objdump SIZE=mipsel-openwrt-linux-musl-size CROSS="mipsel-openwrt-linux-musl-" ARCH="mipsel" ;
make[3]: Entering directory `/home/azureuser/build/openwrt-github/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.16/libprotobuf-c-v1.2.1'
make[3]: *** No targets specified and no makefile found.  Stop.
make[3]: Leaving directory `/home/azureuser/build/openwrt-github/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.16/libprotobuf-c-v1.2.1'
make[2]: *** [/home/azureuser/build/openwrt-github/openwrt/build_dir/target-mipsel_24kec+dsp_musl-1.1.16/libprotobuf-c-v1.2.1/.built] Error 2
make[2]: Leaving directory `/home/azureuser/build/openwrt-github/openwrt/feeds/packages/libs/protobuf-c'
make[1]: *** [package/feeds/packages/protobuf-c/compile] Error 2
make[1]: Leaving directory `/home/azureuser/build/openwrt-github/openwrt'
make: *** [package/protobuf-c/compile] Error 2
```

jsiverskog, 2017-03-16:
Does `make package/libprotobuf-c/compile V=s` work?

I'm not sure what naming standards there are, but I agree that the package name and path should match.

ppaulojr, 2017-03-16:
@jsiverskog no it doesn't work but the PR I made works.

https://github.com/openwrt/packages/issues/4156

hnyman, 2017-05-10:
Not sure where your error comes from. 
Both LEDE and Openwrt buildbots build the package quite normally and I just tested it myself. Built ok.

```
perus@ub1704:/OwrtLEDE/r7800$ make package/protobuf-c/{clean,compile} V=s
make[1]: Entering directory '/OwrtLEDE/r7800'
make[2]: Entering directory '/OwrtLEDE/r7800/feeds/packages/libs/protobuf-c'
...
touch /OwrtLEDE/r7800/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.protobuf-c_installed
make[2]: Leaving directory '/OwrtLEDE/r7800/feeds/packages/libs/protobuf-c'
make[1]: Leaving directory '/OwrtLEDE/r7800'

perus@ub1704:/OwrtLEDE/r7800$ ls bin/packages/arm_cortex-a15_neon-vfpv4/packages/libprotobuf*
bin/packages/arm_cortex-a15_neon-vfpv4/packages/libprotobuf-c_v1.2.1_arm_cortex-a15_neon-vfpv4.ipk
```

worksforme

closed 2017-05-10 by hnyman