openwrt/packages #4159: unbound: /etc/init.d/unbound returns "cp: unrecognized option: u"

vollkommenheit, 2017-03-18:
Since at least unbound_1.6.1-2, I've been getting "cp: unrecognized option: u" when executing "/etc/init.d/unbound restart" which seems attributable to the "cp -u $UNBOUND_KEYFILE.keep $UNBOUND_KEYFILE" line in /usr/lib/unbound/unbound.sh

EricLuehrsen, 2017-03-21:
Did you install a different set of utilities?
```
# cp --help
BusyBox v1.26.2 () multi-call binary.

Usage: cp [OPTIONS] SOURCE... DEST

Copy SOURCE(s) to DEST

	-a	Same as -dpR
	-R,-r	Recurse
	-d,-P	Preserve symlinks (default if -R)
	-L	Follow all symlinks
	-H	Follow symlinks on command line
	-p	Preserve file attributes if possible
	-f	Overwrite
	-i	Prompt before overwrite
	-l,-s	Create (sym)links
	-u	Copy only newer files
```
Also with gnu core-utils.
```
‘-u’
‘--update’
    Do not copy a non-directory that has an existing destination with the same or newer modification timestamp.
```

hnyman, 2017-03-21:
> > Since at least unbound_1.6.1-2, I've been getting "cp: unrecognized option: u"

> Did you install a different set of utilities?

The reason is probably old busybox in the stagnant Openwrt tree...

support for "cp -u" has been added to busybox in May 2016 so that it is found from Busybox 1.25 and later. LEDE currently uses 1.26.2, but the Openwrt tree still uses 1.24.2 so there is no support for "cp -u" in Openwrt.

https://git.busybox.net/busybox/commit/coreutils/cp.c?id=877dedb8251be47b3614a371434081ae9b7b358b

https://github.com/openwrt/openwrt/blob/master/package/utils/busybox/Makefile
https://github.com/lede-project/source/blob/master/package/utils/busybox/Makefile

EricLuehrsen, 2017-03-21:
@hnyman, avoiding the politics if it, what is the expected support plan? OpenWrt 15.05 is the last official release and other than DS record for 2017, Unbound is left untouched. LEDE 17.01 represents a lot of forward progress, and backwards compatibility to trunk self-builds on OpenWrt is going to get harder. I want to do the "right" thing but its not easy.

hnyman, 2017-03-21:
> avoiding the politics if it, what is the expected support plan? 

I don't know any expected support plan. So far both Luci and packages feeds have mainly stayed compatible with the progressing LEDE and the stagnant Openwrt, but it is getting harder and harder as the age gap between the two distros grow. Just like this minor command difference shows :-( 

There have been some earlier breakages that have been fixed in packages, (e.g. related to openssl sslv3 feature handling etc.), package Makefiles in general, cake qdisc, SSL cert generation etc. But the differences in kernel, musl, busybox etc. core components will start breaking things sooner or later in a more major way.

It would be great if you could figure out a way to fix unbound to work also with Openwrt. (a log warning to install gnu coreutils cp, if "cp -u" fails? or something like that?)

As LEDE is being actively developed, and very likely the development after a possible merge (which may happen, or not) would continue from LEDE, I feel that the compatibility with it is more crucial than Openwrt.

I would say that currently only LEDE and LEDE 17.01 are actively developed/maintained, while Openwrt master and 15.05 slowly rot away. 15.05 is mainly unmaintained, (e.g. the CC15.05 source repo has got its git/svn revision detection broken, for which I wrote a PR in October, which PR is still unmerged). I have stopped building CC15.05 myself. I still build Openwrt master for my WNDR3700 community build maybe once per month just see if all the packages in my old build still build ok with it.

vollkommenheit, 2017-03-21:
FWIW, I'm using a trunk build of openwrt DESIGNATED DRIVER from December 2016, which still has BusyBox v1.24.2.

EricLuehrsen, 2017-03-22:
@vollkommenheit please confirm #4179 works for you.

vollkommenheit, 2017-03-22:
Yup, works great here.

closed 2017-03-22 by hnyman