openwrt/packages #4162: <Python>There aren't package/feeds/packages/python/host/compile

JefferyKu, 2017-03-18:
touch /home/parallels/BPI-OpenWRT/build_dir/host/expat-2.1.0/.configuredmake -j1 -C /home/parallels/BPI-OpenWRT/build_dir/host/expat-2.1.0  make[4]: Entering directory '/home/parallels/BPI-OpenWRT/build_dir/host/expat-2.1.0'/usr/bin/env bash ./libtool --silent --mode=compile gcc -I./lib -I. -O2 -I/home/parallels/BPI-OpenWRT/staging_dir/host/include -I/home/parallels/BPI-OpenWRT/staging_dir/host/usr/include -Wall -Wmissing-prototypes -Wstrict-prototypes -fexceptions  -I/home/parallels/BPI-OpenWRT/staging_dir/host/include -I/home/parallels/BPI-OpenWRT/staging_dir/host/usr/include -DHAVE_EXPAT_CONFIG_H -o lib/xmlparse.lo -c lib/xmlparse.c/usr/bin/env bash ./libtool --silent --mode=compile gcc -I./lib -I. -O2 -I/home/parallels/BPI-OpenWRT/staging_dir/host/include -I/home/parallels/BPI-OpenWRT/staging_dir/host/usr/include -Wall -Wmissing-prototypes -Wstrict-prototypes -fexceptions  -I/home/parallels/BPI-OpenWRT/staging_dir/host/include -I/home/parallels/BPI-OpenWRT/staging_dir/host/usr/include -DHAVE_EXPAT_CONFIG_H -o lib/xmltok.lo -c lib/xmltok.c/usr/bin/env bash ./libtool --silent --mode=compile gcc -I./lib -I. -O2 -I/home/parallels/BPI-OpenWRT/staging_dir/host/include -I/home/parallels/BPI-OpenWRT/staging_dir/host/usr/include -Wall -Wmissing-prototypes -Wstrict-prototypes -fexceptions  -I/home/parallels/BPI-OpenWRT/staging_dir/host/include -I/home/parallels/BPI-OpenWRT/staging_dir/host/usr/include -DHAVE_EXPAT_CONFIG_H -o lib/xmlrole.lo -c lib/xmlrole.c/usr/bin/env bash ./libtool --silent --mode=link gcc -I./lib -I. -O2 -I/home/parallels/BPI-OpenWRT/staging_dir/host/include -I/home/parallels/BPI-OpenWRT/staging_dir/host/usr/include -Wall -Wmissing-prototypes -Wstrict-prototypes -fexceptions  -I/home/parallels/BPI-OpenWRT/staging_dir/host/include -I/home/parallels/BPI-OpenWRT/staging_dir/host/usr/include -DHAVE_EXPAT_CONFIG_H -no-undefined -version-info 7:0:6 -rpath /home/parallels/BPI-OpenWRT/staging_dir/host/lib -L/home/parallels/BPI-OpenWRT/staging_dir/host/lib -L/home/parallels/BPI-OpenWRT/staging_dir/host/usr/lib -o libexpat.la lib/xmlparse.lo lib/xmltok.lo lib/xmlrole.lox86_64-linux-gnu-ar: `u' modifier ignored since `D' is the default (see `U')gcc -I./lib -I. -O2 -I/home/parallels/BPI-OpenWRT/staging_dir/host/include -I/home/parallels/BPI-OpenWRT/staging_dir/host/usr/include -Wall -Wmissing-prototypes -Wstrict-prototypes -fexceptions  -I/home/parallels/BPI-OpenWRT/staging_dir/host/include -I/home/parallels/BPI-OpenWRT/staging_dir/host/usr/include -DHAVE_EXPAT_CONFIG_H -o xmlwf/xmlwf.o -c xmlwf/xmlwf.cgcc -I./lib -I. -O2 -I/home/parallels/BPI-OpenWRT/staging_dir/host/include -I/home/parallels/BPI-OpenWRT/staging_dir/host/usr/include -Wall -Wmissing-prototypes -Wstrict-prototypes -fexceptions  -I/home/parallels/BPI-OpenWRT/staging_dir/host/include -I/home/parallels/BPI-OpenWRT/staging_dir/host/usr/include -DHAVE_EXPAT_CONFIG_H -o xmlwf/xmlfile.o -c xmlwf/xmlfile.cgcc -I./lib -I. -O2 -I/home/parallels/BPI-OpenWRT/staging_dir/host/include -I/home/parallels/BPI-OpenWRT/staging_dir/host/usr/include -Wall -Wmissing-prototypes -Wstrict-prototypes -fexceptions  -I/home/parallels/BPI-OpenWRT/staging_dir/host/include -I/home/parallels/BPI-OpenWRT/staging_dir/host/usr/include -DHAVE_EXPAT_CONFIG_H -o xmlwf/codepage.o -c xmlwf/codepage.cgcc -I./lib -I. -O2 -I/home/parallels/BPI-OpenWRT/staging_dir/host/include -I/home/parallels/BPI-OpenWRT/staging_dir/host/usr/include -Wall -Wmissing-prototypes -Wstrict-prototypes -fexceptions  -I/home/parallels/BPI-OpenWRT/staging_dir/host/include -I/home/parallels/BPI-OpenWRT/staging_dir/host/usr/include -DHAVE_EXPAT_CONFIG_H -o xmlwf/unixfilemap.o -c xmlwf/unixfilemap.c/usr/bin/env bash ./libtool --silent --mode=link gcc -I./lib -I. -O2 -I/home/parallels/BPI-OpenWRT/staging_dir/host/include -I/home/parallels/BPI-OpenWRT/staging_dir/host/usr/include -Wall -Wmissing-prototypes -Wstrict-prototypes -fexceptions  -I/home/parallels/BPI-OpenWRT/staging_dir/host/include -I/home/parallels/BPI-OpenWRT/staging_dir/host/usr/include -DHAVE_EXPAT_CONFIG_H -L/home/parallels/BPI-OpenWRT/staging_dir/host/lib -L/home/parallels/BPI-OpenWRT/staging_dir/host/usr/lib -o xmlwf/xmlwf xmlwf/xmlwf.o xmlwf/xmlfile.o xmlwf/codepage.o xmlwf/unixfilemap.o libexpat.lamake[4]: Leaving directory '/home/parallels/BPI-OpenWRT/build_dir/host/expat-2.1.0'touch /home/parallels/BPI-OpenWRT/build_dir/host/expat-2.1.0/.builtmake -C /home/parallels/BPI-OpenWRT/build_dir/host/expat-2.1.0 installmake[4]: Entering directory '/home/parallels/BPI-OpenWRT/build_dir/host/expat-2.1.0'/usr/bin/env bash ./conftools/mkinstalldirs /home/parallels/BPI-OpenWRT/staging_dir/host/lib /home/parallels/BPI-OpenWRT/staging_dir/host/include /home/parallels/BPI-OpenWRT/staging_dir/host/lib/pkgconfig/usr/bin/env bash ./libtool --mode=install /usr/bin/install -c libexpat.la /home/parallels/BPI-OpenWRT/staging_dir/host/lib/libexpat.laOpenWrt-libtool: install: /usr/bin/install -c .libs/libexpat.so.1.6.0 /home/parallels/BPI-OpenWRT/staging_dir/host/lib/libexpat.so.1.6.0OpenWrt-libtool: install: (cd /home/parallels/BPI-OpenWRT/staging_dir/host/lib && { ln -s -f libexpat.so.1.6.0 libexpat.so.1 || { rm -f libexpat.so.1 && ln -s libexpat.so.1.6.0 libexpat.so.1; }; })OpenWrt-libtool: install: (cd /home/parallels/BPI-OpenWRT/staging_dir/host/lib && { ln -s -f libexpat.so.1.6.0 libexpat.so || { rm -f libexpat.so && ln -s libexpat.so.1.6.0 libexpat.so; }; })OpenWrt-libtool: install: /usr/bin/install -c .libs/libexpat.lai /home/parallels/BPI-OpenWRT/staging_dir/host/lib/libexpat.laOpenWrt-libtool: install: /usr/bin/install -c .libs/libexpat.a /home/parallels/BPI-OpenWRT/staging_dir/host/lib/libexpat.aOpenWrt-libtool: install: chmod 644 /home/parallels/BPI-OpenWRT/staging_dir/host/lib/libexpat.aOpenWrt-libtool: install: x86_64-linux-gnu-ranlib /home/parallels/BPI-OpenWRT/staging_dir/host/lib/libexpat.aOpenWrt-libtool: finish: PATH="/home/parallels/BPI-OpenWRT/staging_dir/host/bin:/home/parallels/BPI-OpenWRT/staging_dir/toolchain-arm_cortex-a8+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/bin:/home/parallels/BPI-OpenWRT/staging_dir/toolchain-arm_cortex-a8+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/bin:/home/parallels/BPI-OpenWRT/staging_dir/host/bin:/home/parallels/BPI-OpenWRT/staging_dir/toolchain-arm_cortex-a8+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/bin:/home/parallels/BPI-OpenWRT/staging_dir/host/bin:/home/parallels/BPI-OpenWRT/staging_dir/host/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/sbin" ldconfig -n /home/parallels/BPI-OpenWRT/staging_dir/host/lib----------------------------------------------------------------------Libraries have been installed in:   /home/parallels/BPI-OpenWRT/staging_dir/host/libIf you ever happen to want to link against installed librariesin a given directory, LIBDIR, you must either use libtool, andspecify the full pathname of the library, or use the `-LLIBDIR'flag during linking and do at least one of the following:   - add LIBDIR to the `LD_LIBRARY_PATH' environment variable     during execution   - add LIBDIR to the `LD_RUN_PATH' environment variable     during linking   - use the `-Wl,-rpath -Wl,LIBDIR' linker flag   - have your system administrator add LIBDIR to `/etc/ld.so.conf'See any operating system documentation about shared libraries formore information, such as the ld(1) and ld.so(8) manual pages.----------------------------------------------------------------------for FN in ./lib/expat.h ./lib/expat_external.h ; do /usr/bin/install -c -m 644 $FN /home/parallels/BPI-OpenWRT/staging_dir/host/include ; done/usr/bin/install -c -m 644 expat.pc /home/parallels/BPI-OpenWRT/staging_dir/host/lib/pkgconfig/expat.pc/usr/bin/env bash ./conftools/mkinstalldirs /home/parallels/BPI-OpenWRT/staging_dir/host/bin /home/parallels/BPI-OpenWRT/staging_dir/host/share/man/man1/usr/bin/env bash ./libtool --mode=install /usr/bin/install -c xmlwf/xmlwf /home/parallels/BPI-OpenWRT/staging_dir/host/bin/xmlwfOpenWrt-libtool: install: /usr/bin/install -c xmlwf/.libs/xmlwf /home/parallels/BPI-OpenWRT/staging_dir/host/bin/xmlwf/usr/bin/install -c -m 644 ./doc/xmlwf.1 /home/parallels/BPI-OpenWRT/staging_dir/host/share/man/man1make[4]: Leaving directory '/home/parallels/BPI-OpenWRT/build_dir/host/expat-2.1.0'mkdir -p /home/parallels/BPI-OpenWRT/staging_dir/host/stamptouch /home/parallels/BPI-OpenWRT/build_dir/host/expat-2.1.0/.builttouch /home/parallels/BPI-OpenWRT/staging_dir/host/stamp/.expat_installedmake[3]: **Leaving dir [... cut ...] mp/.package_compile] Error 2make[1]: Leaving directory '/home/parallels/BPI-OpenWRT'/home/parallels/BPI-OpenWRT/include/toplevel.mk:181: recipe for target 'world' failedmake: *** [world] Error 2

I check this directory, package/feed/packages/python/host/compile. I had made my menuconfig including python and python3. However, there are still problem on this. How to fix it? I am git newbie also. Please kindly teach me. Thanks.

What I clone is this repo https://github.com/BPI-SINOVOIP/BPI-OpenWRT.

hnyman, 2017-03-18:
> What I clone is this repo https://github.com/BPI-SINOVOIP/BPI-OpenWRT.

You are pretty much on your own. That non-official repo contains two year old un-updated code.

> > based on OpenWRT15.05-rc3 

You should be compiling from Openwrt or LEDE master (or Openwrt 15.05 branch after 15.05.1 release). Packages in the package feed evolve along the main source repos and trying to use current packages with a 2-year old main code will not work.

closed 2017-03-18 by hnyman

JefferyKu, 2017-03-19:
Hi,
I use Banana pi this board. This source is from a blog's introduction. I don't know there are two years no more update. Thanks your reply first. 
So, do I give up this Banana pi repo and use official openwrt and LEDE master? I had read some articles about basic git command and its operation method & principle. But, I am new on this tool. Is it possible to combine the official openwrt package to BPI-SINOVOIP/BPI-OpenWRT. Currently, I got stuck on python this package. How to merge official package, /openwrt/lang/python (https://github.com/openwrt/packages/tree/master/lang/python)? It is my first time to use git. What I think is possible to be incomplete.

diizzyy, 2017-03-21:
There's no support for the Banana Pi board in upstream (LEDE or OpenWRT repo).