openwrt/packages #4182: package opencv compile failed

taeasy, 2017-03-22:
Maintainer: @ jazzgs
Environment:  X86_64, LEDE 17.01

Description:

make[4]: Leaving directory '/home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0'
Install the project...
-- Install configuration: "Release"
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/share/OpenCV/3rdparty/lib/libippicv.a
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/cvconfig.h
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/opencv_modules.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/lib/pkgconfig/opencv.pc
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/share/OpenCV/OpenCVConfig.cmake
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/share/OpenCV/OpenCVConfig-version.cmake
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/share/OpenCV/OpenCVModules.cmake
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/share/OpenCV/OpenCVModules-release.cmake
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv/cv.h
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv/cv.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv/cvaux.h
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv/cvaux.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv/cvwimage.h
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv/cxcore.h
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv/cxcore.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv/cxeigen.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv/cxmisc.h
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv/highgui.h
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv/ml.h
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/opencv.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/lib/libopencv_core.so.3.1.0
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/lib/libopencv_core.so.3.1
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/lib/libopencv_core.so
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/block.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/border_interpolate.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/color.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/common.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/datamov_utils.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/dynamic_smem.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/emulation.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/filters.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/funcattrib.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/functional.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/limits.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/reduce.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/saturate_cast.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/scan.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/simd_functions.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/transform.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/type_traits.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/utility.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/vec_distance.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/vec_math.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/vec_traits.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/warp.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/warp_reduce.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/warp_shuffle.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/detail/color_detail.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/detail/reduce.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/detail/reduce_key_val.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/detail/transform_detail.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/detail/type_traits_detail.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda/detail/vec_distance_detail.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/affine.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/base.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/bufferpool.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/core.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda.inl.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda_stream_accessor.hpp
-- Installing: /home/main/lede/build_dir/target-x86_64_musl-1.1.16/opencv-3.1.0/ipkg-install/usr/include/opencv2/core/cuda_types.hpp
[... 159 lines cut ...]
[... cut ...]
make[2]: *** [/home/main/lede/bin/packages/x86_64/packages/opencv_3.1.0-2_x86_64.ipk] Error 1
make[2]: Leaving directory '/home/main/lede/feeds/packages/libs/opencv'
package/Makefile:105: recipe for target 'package/feeds/packages/opencv/compile' failed
make[1]: *** [package/feeds/packages/opencv/compile] Error 2
make[1]: Leaving directory '/home/main/lede'
/home/main/lede/include/toplevel.mk:197: recipe for target 'package/opencv/compile' failed
make: *** [package/opencv/compile] Error 2
main@debian:~/lede$

ckq95995, 2017-06-09:
I have the exact same problem. Anyone can give a tip?

bazilxp, 2017-11-20:
Add to depends section        +glib2 to the end:)

dibdot, 2018-08-17:
Please retry with 18.06.

closed 2018-08-17 by dibdot