openwrt/packages #4191: `shadowsocks-libev:` 2.6.1 to 3.0.5

shounak-de, 2017-03-23:
-------------------------------

Maintainer: @diizzyy @aa65535
Environment: OpenWRT (need in downstream [entware-ng](https://github.com/Entware-ng/Entware-ng))

Description:

please update shadowsocks-libev to current 3.0.5 version.

https://github.com/shadowsocks/shadowsocks-libev/releases

thanks.

diizzyy, 2017-03-24:
Don't use it so I can't test....
As a sidenote, there's another repo/fork for some reason...
https://github.com/fuckshadows

steve3d, 2017-04-10:
you can not test it because you don't need it, but I have to say, there are SO many people still need it, so please update this to latest version of 3.0.6.

Thanks.

joopdorresteijn, 2017-05-06:
Yep! Would be really happy with are update. My router doesn't have the space to install GCC...

LukaszWolinski, 2017-05-08:
I am in the same situation @joopdorresteijn . I hope we can get it solved fast.

yousong, 2017-05-08:
Just out of curiosity, are there some nice new features not available in 2.6.1?

joopdorresteijn, 2017-05-08:
@yousong 

Stuff I want from changelog in **bold**. Esp. the cyphers and dropping OTA has been applied in various clients downstream, making version 2.6 obsolete. Steps in the good direction technology wise, speeding up and obfuscating at same time.

1. Fix a bug with AEAD ciphers.
2. Refine ACL support by @blackgear.
**3. Fix a bug of TCP Fast Open in ss-redir.**
4. Add CMake files by @wenerme.
**5. Support TCP Fast Open in ss-redir by @lqs.**
6. Support TOS/DESCP in ss-redir by @sduponch.
7. Refine MPTCP by @sduponch.
8. Replace nonce cache with a ping-pong bloom filter.
9. Add session key for AEAD. (SIP007)
10. Fix a crashe when using stream ciphers.
11. Fix a protocol bug in AEAD ciphers. (SIP004)
12. Allow setting keys directly. (SIP006)
13. Drop dependencies of OpenSSL and PolarSSL.
**14. Deprecate OTA (One-Time-Auth).**
**15. **Add new ciphers for SIP004: aes-128-gcm, aes-192-gcm, aes-256-gcm,****
16. chacha20-poly1305 and chacha20-ietf-poly1305.
17. Refine SIP003 to support standalone mode of obfsproxy.
18. Refine the project structure.
19. Refine SIP003 plugin support.
20.

yousong, 2017-05-12:
Hi, I just opened a PR to bump the version to 3.0.6.  Please review and test if it works for you.

I only occasionally need to use ss-local which is not part of the shadowsocks-libev package which is already very big for my device.  So I created my own feed and build \[1,2].

 \[1] https://github.com/yousong/waller/blob/master/shadowsocks-libev/Makefile
 \[2] https://github.com/yousong/build-scripts/blob/master/build-shadowsocks-libev.sh

joopdorresteijn, 2017-05-15:
Thanks Yousong, as I don't have a build environment for this architecture, I will need to wait until it's in openwrt as downstream https://github.com/Entware-ng/Entware-ng/issues/639 will only make a build once it's in openwrt. Cool that you did it~!

yousong, 2017-05-22:
FYI, The pull request to update shadowsocks-libev to 3.0.6 has just been pushed a moment ago.

closed 2017-05-22 by yousong

yousong, 2017-06-26:
Hi, @joopdorresteijn , @LukaszWolinski , @shounak-de , @diizzyy , @steve3d , 

Not sure how you guys are using shadowsocks-libev in this repo, but I am rewriting it trying to bring other components to the party.  Please give https://github.com/openwrt/packages/pull/4513 a try and see if it can fulfil your needs.  Thanks.

shounak-de, 2017-06-26:
I very glad that it addresses #4455

la88962, 2017-07-17:
Please include ss-server

yousong, 2017-07-22:
Hi, I just opened a pr for the rewritten version of `luci-app-shadowsocks-libev` (openwrt/luci/pull/1242) and I think it should be now in a ready to ship state.  It would be great if you can give it a try and provide your feedback there, e.g. about the phrasing and wording etc.