openwrt/packages #4198: python host comiple failed

roboidler, 2017-03-25:
dear all,

after i feeds update -a install -a,  python host compile failed..

openwrt trunk

running build_scripts
creating build/scripts-2.7
copying /home/roboidler/workspace/openwrt/trunk/openwrt/build_dir/target-mips_34kc_musl-1.1.14/host/Python-2.7.13/Tools/scripts/pydoc -> build/scripts-2.7
copying /home/roboidler/workspace/openwrt/trunk/openwrt/build_dir/target-mips_34kc_musl-1.1.14/host/Python-2.7.13/Tools/scripts/idle -> build/scripts-2.7
copying /home/roboidler/workspace/openwrt/trunk/openwrt/build_dir/target-mips_34kc_musl-1.1.14/host/Python-2.7.13/Tools/scripts/2to3 -> build/scripts-2.7
copying /home/roboidler/workspace/openwrt/trunk/openwrt/build_dir/target-mips_34kc_musl-1.1.14/host/Python-2.7.13/Lib/smtpd.py -> build/scripts-2.7
/usr/bin/install -c -m 644 ./Tools/gdb/libpython.py python-gdb.py
make[4]: Leaving directory `/home/roboidler/workspace/openwrt/trunk/openwrt/build_dir/target-mips_34kc_musl-1.1.14/host/Python-2.7.13'
touch /home/roboidler/workspace/openwrt/trunk/openwrt/build_dir/target-mips_34kc_musl-1.1.14/host/Python-2.7.13/.built
make -C /home/roboidler/workspace/openwrt/trunk/openwrt/build_dir/target-mips_34kc_musl-1.1.14/host/Python-2.7.13 install
make[4]: Entering directory `/home/roboidler/workspace/openwrt/trunk/openwrt/build_dir/target-mips_34kc_musl-1.1.14/host/Python-2.7.13'
/usr/bin/install -c python /bin/python2.7
/usr/bin/install: cannot remove '/bin/python2.7': Permission denied
make[4]: *** [altbininstall] Error 1

Regards,

r

diizzyy, 2017-03-26:
Not a fix but you can't use the LEDE trunk instead which is much newer?

hnyman, 2017-03-26:
Based on `target-mips_34kc_musl-1.1.14` you are using the abandoned git.openwrt.org sources, or otherwise quite outdated. (musl is 1.1.16 in both LEDE and Openwrt trunk)

Python package Makefiles are incompatible with so old sources.

Use https://github.com/openwrt/openwrt for current Openwrt sources or 	
https://git.lede-project.org/source.git for the current LEDE sources

closed 2017-03-26 by hnyman