openwrt/packages #4202: Mutt for BananaPI (sunxi)

donaufisch66, 2017-03-26:
Hello,
I am looking for the package "Mutt" for "SUNXI" (BananaPI). I can find the package for other hardware but not for mine.
How can I use Mut from a different package or how can I make a new package for my SUNXI Platform?
MUTT: www.mutt.org
SUNXI Packages:
http://downloads.openwrt.org/chaos_calmer/15.05/sunxi/generic/packages/packages

diizzyy, 2017-03-26:
You need to either build yourself or use a trunk/master depending if you want to run OpenWRT or LEDE.

donaufisch66, 2017-03-27:
trunk doesnt work. I am not able to build a package.

yousong, 2017-03-28:
Checking with the git log, mutt was introduced into this repo at 2015-11-18, before OpenWrt CC-15.05.  I would recommend switching to LEDE-17.01

closed 2018-01-19 by yousong