openwrt/packages #4210: lighttpd: PKG_CONFIG_DEPENDS missing

nwf, 2017-03-27:
Maintainer: @MikePetullo 
Environment: Kirkwood, LEDE HEAD.

Toggling `CONFIG_PACKAGE_lighttpd-mod-authn_gssapi` from `n` to `y` in an existing build gave rise to

```
make[3]: Entering directory '/tank/openwrt/scratch/builder-webserver/feeds/packages/net/lighttpd'
mkdir -p /tank/openwrt/scratch/builder-webserver/staging_dir/target-arm_xscale_musl_eabi/root-kirkwood/stamp
SHELL= flock /tank/openwrt/scratch/builder-webserver/tmp/.root-copy.flock -c 'cp -fpR /tank/openwrt/scratch/builder-webserver/build_dir/target-arm_xscale_musl_eabi/lighttpd-1.4.45/.pkgdir/lighttpd/. /tank/openwrt/scratch/builder-webserver/staging_dir/target-arm_xscale_musl_eabi/root-kirkwood/'
touch /tank/openwrt/scratch/builder-webserver/staging_dir/target-arm_xscale_musl_eabi/root-kirkwood/stamp/.lighttpd_installed
mkdir -p /tank/openwrt/scratch/builder-webserver/bin/targets/kirkwood/generic/packages /tank/openwrt/scratch/builder-webserver/build_dir/target-arm_xscale_musl_eabi/lighttpd-1.4.45/ipkg-arm_xscale/lighttpd-mod-authn_gssapi/CONTROL /tank/openwrt/scratch/builder-webserver/staging_dir/target-arm_xscale_musl_eabi/pkginfo
install -d -m0755 /tank/openwrt/scratch/builder-webserver/build_dir/target-arm_xscale_musl_eabi/lighttpd-1.4.45/ipkg-arm_xscale/lighttpd-mod-authn_gssapi/usr/lib/lighttpd
cp -fpR /tank/openwrt/scratch/builder-webserver/build_dir/target-arm_xscale_musl_eabi/lighttpd-1.4.45/ipkg-install/usr/lib/lighttpd/mod_authn_gssapi.so /tank/openwrt/scratch/builder-webserver/build_dir/target-arm_xscale_musl_eabi/lighttpd-1.4.45/ipkg-arm_xscale/lighttpd-mod-authn_gssapi/usr/lib/lighttpd
cp: cannot stat '/tank/openwrt/scratch/builder-webserver/build_dir/target-arm_xscale_musl_eabi/lighttpd-1.4.45/ipkg-install/usr/lib/lighttpd/mod_authn_gssapi.so': No such file or directory
Makefile:197: recipe for target '/tank/openwrt/scratch/builder-webserver/bin/packages/arm_xscale/packages/lighttpd-mod-authn_gssapi_1.4.45-3_arm_xscale.ipk' failed
```

Removing `/tank/openwrt/scratch/builder-webserver/build_dir/target-arm_xscale_musl_eabi/lighttpd-1.4.45` and rebuilding produced success.  I believe this because the lighttpd makefile neglects to define `PKG_CONFIG_DEPENDS`.  I believe, but am not sure, that it would suffice to add `PKG_CONFIG_DEPENDS+=lighttpd-mod-$(1)` to the `BuildPlugin` macro's definition.

gstrauss, 2017-04-16:
As @nwf suggests, it would appear that PKG_CONFIG_DEPENDS is what is needed, but only when selection of a module package results in the need to re-run ./confgure and rebuild lighttpd.  See comment at https://github.com/openwrt/packages/blob/master/net/openvswitch/Makefile#L47

Here is what I plan to test;
```
# trigger rebuild of lighttpd package if module package selection changes
PKG_CONFIG_DEPENDS := \
        CONFIG_LIGHTTPD_SSL \
        CONFIG_PACKAGE_lighttpd-mod-authn_gssapi \
        CONFIG_PACKAGE_lighttpd-mod-authn_ldap \
        CONFIG_PACKAGE_lighttpd-mod-authn_mysql \
        CONFIG_PACKAGE_lighttpd-mod-cml \
        CONFIG_PACKAGE_lighttpd-mod-magnet \
        CONFIG_PACKAGE_lighttpd-mod-mysql_vhost \
        CONFIG_PACKAGE_lighttpd-mod-trigger_b4_dl \
        CONFIG_PACKAGE_lighttpd-mod-webdav
```

nwf, 2017-04-18:
Any reason to do this manually rather than automagically within `BuildPlugin`?  Do some plugins not need to induce reconfiguration?

hnyman, 2017-04-18:
> Do some plugins not need to induce reconfiguration?

I think that only those plugins need to trigger reconfiguration that affect CONFIGURE_ARGS, CONFIGURE_VARS or compile/linking options. Based on Makefile, it is pretty much the list from @gstrauss

gstrauss, 2017-04-18:
> Do some plugins not need to induce reconfiguration?

Most plugins *do not* need to induce reconfiguration.  The plugins that need to induce reconfiguration are those that change the arguments given to `./configure`, e.g. those plugins that depend on external libraries, such as database client libraries, for which there is a --with-xxxxx argument to configure.ac

mhei, 2018-01-08:
@MikePetullo: please have a look whether we can fix this long-standing issue this way. Thnx.

commit 9ca0961d (2018-01-09): lighttpd: fix PKG_CONFIG_DEPENDS (fixes #4210)
commit e7f345f5 (2018-01-09): Merge pull request #5385 from mhei/lighttpd-issue-4210

lighttpd: fix PKG_CONFIG_DEPENDS (fixes #4210)
closed 2018-01-09 by mhei