openwrt/packages #4212: dnsmasq: sequential IP allocation not working

psyborg55, 2017-03-28:
ticked Allocate IP sequentially in DHCP and DNS settings but still got random IP addresses assigned

```
Tue Mar 28 04:06:14 2017 daemon.info dnsmasq-dhcp[16634]: DHCPREQUEST(wlan2-1) 192.168.15.47 2c:d0:5a:76:b5:2e 
Tue Mar 28 04:06:14 2017 daemon.info dnsmasq-dhcp[16634]: DHCPACK(wlan2-1) 192.168.15.47 2c:d0:5a:76:b5:2e  ubuntu
Tue Mar 28 04:08:16 2017 daemon.info dnsmasq-dhcp[16634]: DHCPDISCOVER(wlan0-4) 192.168.9.202 2c:d0:5a:76:b5:2e 
Tue Mar 28 04:08:16 2017 daemon.info dnsmasq-dhcp[16634]: DHCPOFFER(wlan0-4) 192.168.9.202 2c:d0:5a:76:b5:2e 
Tue Mar 28 04:08:16 2017 daemon.info dnsmasq-dhcp[16634]: DHCPREQUEST(wlan0-4) 192.168.9.202 2c:d0:5a:76:b5:2e 
Tue Mar 28 04:08:16 2017 daemon.info dnsmasq-dhcp[16634]: DHCPACK(wlan0-4) 192.168.9.202 2c:d0:5a:76:b5:2e  ubuntu
```

version 2.76-1 r50082

hnyman, 2017-03-28:
dnsmasq is not here in the pckages repo.

Please file the issue in Openwrt main repo.

closed 2017-03-28 by hnyman

psyborg55, 2017-03-28:
whoever saw this before it was closed is familiar with the issue. i will not open it again unless i can open it on trac.