openwrt/packages #4216: ffmpeg: "ERROR: shine not found using pkg-config"

youxiaojie, 2017-03-29:
make[3]: Entering directory '/media/allan/data/lede-git/source/feeds/packages/multimedia/ffmpeg'
rm -f /media/allan/data/lede-git/source/staging_dir/target-mips_24kc_musl/stamp/.ffmpeg.full_installed
( cd /media/allan/data/lede-git/source/build_dir/target-mips_24kc_musl/ffmpeg-full/ffmpeg-3.2.4; CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/media/allan/data/lede-git/source/build_dir/target-mips_24kc_musl/ffmpeg-full/ffmpeg-3.2.4:ffmpeg-3.2.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/media/allan/data/lede-git/source/staging_dir/target-mips_24kc_musl/usr/include -I/media/allan/data/lede-git/source/staging_dir/target-mips_24kc_musl/include -I/media/allan/data/lede-git/source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/usr/include -I/media/allan/data/lede-git/source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/include/fortify -I/media/allan/data/lede-git/source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/include -fpic" LDFLAGS="-L/media/allan/data/lede-git/source/staging_dir/target-mips_24kc_musl/usr/lib -L/media/allan/data/lede-git/source/staging_dir/target-mips_24kc_musl/lib -L/media/allan/data/lede-git/source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/usr/lib -L/media/allan/data/lede-git/source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/lib -znow -zrelro" ./configure --enable-cross-compile --cross-prefix="mips-openwrt-linux-musl-" --arch="mips" --target-os=linux --prefix="/usr" --pkg-config="pkg-config" --enable-shared --enable-static --enable-pthreads --enable-zlib --disable-doc --disable-debug --disable-dxva2 --disable-lzma --disable-vaapi --disable-vda --disable-vdpau --disable-outdevs --disable-altivec --disable-vsx --disable-power8 --disable-amd3dnow --disable-amd3dnowext --disable-mmx --disable-mmxext --disable-sse --disable-sse2 --disable-sse3 --disable-ssse3 --disable-sse4 --disable-sse42 --disable-avx --disable-xop --disable-fma3 --disable-fma4 --disable-avx2 --disable-aesni --disable-armv5te --disable-armv6 --disable-armv6t2 --disable-inline-asm --disable-mipsdsp --disable-mipsdspr2 --disable-mipsfpu --disable-msa --disable-mmi --disable-fast-unaligned --disable-runtime-cpudetect --disable-yasm --enable-libopus --enable-small --enable-libshine )
ERROR: shine not found using pkg-config

If you think configure made a mistake, make sure you are using the latest
version from Git.  If the latest version fails, report the problem to the
ffmpeg-user@ffmpeg.org mailing list or IRC #ffmpeg on irc.freenode.net.
Include the log file "config.log" produced by configure as this will help
solve the problem.
Makefile:700: recipe for target '/media/allan/data/lede-git/source/build_dir/target-mips_24kc_musl/ffmpeg-full/ffmpeg-3.2.4/.configured_f78b6842b4d5492e928f33d107540819' failed
make[3]: *** [/media/allan/data/lede-git/source/build_dir/target-mips_24kc_musl/ffmpeg-full/ffmpeg-3.2.4/.configured_f78b6842b4d5492e928f33d107540819] Error 1
make[3]: Leaving directory '/media/allan/data/lede-git/source/feeds/packages/multimedia/ffmpeg'
package/Makefile:109: recipe for target 'package/feeds/packages/ffmpeg/compile' failed
make[2]: *** [package/feeds/packages/ffmpeg/compile] Error 2
make[2]: Leaving directory '/media/allan/data/lede-git/source'

following is last several lines of config.log full file is too big to be pasted here.

check_func mprotect
check_ld cc
check_cc
BEGIN /tmp/ffconf.xbKuqFVg.c
    1	extern int mprotect();
    2	int main(void){ mprotect(); }
END /tmp/ffconf.xbKuqFVg.c
mips-openwrt-linux-musl-gcc -D_ISOC99_SOURCE -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_POSIX_C_SOURCE=200112 -D_XOPEN_SOURCE=600 -DPIC -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/media/allan/data/lede-git/source/build_dir/target-mips_24kc_musl/ffmpeg-full/ffmpeg-3.2.4:ffmpeg-3.2.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/media/allan/data/lede-git/source/staging_dir/target-mips_24kc_musl/usr/include -I/media/allan/data/lede-git/source/staging_dir/target-mips_24kc_musl/include -I/media/allan/data/lede-git/source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/usr/include -I/media/allan/data/lede-git/source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/include/fortify -I/media/allan/data/lede-git/source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/include -fpic -std=c99 -fomit-frame-pointer -fPIC -c -o /tmp/ffconf.CKJ0emqN.o /tmp/ffconf.xbKuqFVg.c
mips-openwrt-linux-musl-gcc -L/media/allan/data/lede-git/source/staging_dir/target-mips_24kc_musl/usr/lib -L/media/allan/data/lede-git/source/staging_dir/target-mips_24kc_musl/lib -L/media/allan/data/lede-git/source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/usr/lib -L/media/allan/data/lede-git/source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/lib -znow -zrelro -Wl,--as-needed -Wl,-z,noexecstack -o /tmp/ffconf.qtN9GqBp /tmp/ffconf.CKJ0emqN.o
check_func_headers time.h nanosleep
check_ld cc
check_cc
BEGIN /tmp/ffconf.xbKuqFVg.c
    1	#include <time.h>
    2	#include <stdint.h>
    3	long check_nanosleep(void) { return (long) nanosleep; }
    4	int main(void) { int ret = 0;
    5	 ret |= ((intptr_t)check_nanosleep) & 0xFFFF;
    6	return ret; }
END /tmp/ffconf.xbKuqFVg.c
mips-openwrt-linux-musl-gcc -D_ISOC99_SOURCE -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_POSIX_C_SOURCE=200112 -D_XOPEN_SOURCE=600 -DPIC -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/media/allan/data/lede-git/source/build_dir/target-mips_24kc_musl/ffmpeg-full/ffmpeg-3.2.4:ffmpeg-3.2.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/media/allan/data/lede-git/source/staging_dir/target-mips_24kc_musl/usr/include -I/media/allan/data/lede-git/source/staging_dir/target-mips_24kc_musl/include -I/media/allan/data/lede-git/source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/usr/include -I/media/allan/data/lede-git/source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/include/fortify -I/media/allan/data/lede-git/source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/include -fpic -std=c99 -fomit-frame-pointer -fPIC -c -o /tmp/ffconf.CKJ0emqN.o /tmp/ffconf.xbKuqFVg.c
mips-openwrt-linux-musl-gcc -L/media/allan/data/lede-git/source/staging_dir/target-mips_24kc_musl/usr/lib -L/media/allan/data/lede-git/source/staging_dir/target-mips_24kc_musl/lib -L/media/allan/data/lede-git/source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/usr/lib -L/media/allan/data/lede-git/source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/lib -znow -zrelro -Wl,--as-needed -Wl,-z,noexecstack -o /tmp/ffconf.qtN9GqBp /tmp/ffconf.CKJ0emqN.o
check_func sched_getaffinity
check_ld cc
check_cc
BEGIN /tmp/ffconf.xbKuqFVg.c
    1	extern int sched_getaffinity();
    2	int main(void){ sched_getaffinity(); }
END /tmp/ffconf.xbKuqFVg.c
mips-openwrt-linux-musl-gcc -D_ISOC99_SOURCE -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -D_POSIX_C_SOURCE=200112 -D_XOPEN_SOURCE=600 -DPIC -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/media/allan/data/lede-git/source/build_dir/target-mips_24kc_musl/ffmpeg-full/ffmpeg-3.2.4:ffmpeg-3.2.4 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/media/allan/data/lede-git/source/staging_dir/target-mips_24kc_musl/usr/include -I/media/allan/data/lede-git/source/staging_dir/target-mips_24kc_musl/include -I/media/allan/data/lede-git/source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/usr/include -I/media/allan/data/lede-git/source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/include/fortify -I/media/allan/data/lede-git/source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/include -fpic -std=c99 -fomit-frame-pointer -fPIC -c -o /tmp/ffconf.CKJ0emqN.o /tmp/ffconf.xbKuqFVg.c
[... cut ...]
mips-openwrt-linux-musl-gcc -L/media/allan/data/lede-git/source/staging_dir/target-mips_24kc_musl/usr/lib -L/media/allan/data/lede-git/source/staging_dir/target-mips_24kc_musl/lib -L/media/allan/data/lede-git/source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/usr/lib -L/media/allan/data/lede-git/source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/lib -znow -zrelro -Wl,--as-needed -Wl,-z,noexecstack -I/media/allan/data/lede-git/source/staging_dir/target-mips_24kc_musl/usr/include/opus -L/media/allan/data/lede-git/source/staging_dir/target-mips_24kc_musl/usr/lib -o /tmp/ffconf.qtN9GqBp /tmp/ffconf.CKJ0emqN.o -lopus -lm -lbz2 -lz -pthread
check_pkg_config s

thess, 2017-03-29:
Sometimes the obvious escapes us... Be sure you have the **shine** package installed. The last time I ran into a similar problem something like:

`$ ./scripts/feeds install shine`

fixed it.

Otherwise, since the buildbots have no problem and I cannot easily reproduce the problem, I can only conclude you have a  non-standard environment and you need to supply more info. The output from:

`$ ./scripts/diffconfig.sh`

would be most helpful as well as any other special setups you are using.

youxiaojie, 2017-03-29:
I run ./scripts/feeds update -a
./script/feeds instll -a
several times......

thess, 2017-03-29:
Once again - please post output from **diffconfig.sh** as requested.

What does: `$ make package/shine/{clean,compile} V=s` produce after the ffmpeg build failure?
Does the directory `staging_dir/target-mips_24kc_musl/usr/include/shine` exist?

diizzyy, 2017-06-02:
@thess 
Close this due to reporter timeout?

closed 2017-06-02 by thess