openwrt/packages #4236: mwan3 ncm / wwan timeing problem?

ThomasCr, 2017-04-04:
Hi,

I run LEDE 17.01 with mwan3 package 2.4-5 from [snapshots](https://downloads.lede-project.org/snapshots/packages/mipsel_74kc/packages/mwan3_2.4-5_all.ipk) on a Archer C7.

My Problem is, that mwan3 hotplug script is not detect, that me wwan connection is up. Only after running the follow command, it works:

`ACTION=ifup INTERFACE=wwan DEVICE=wwan0 sh /etc/hotplug.d/iface/15-mwan3`

I have a slow dsl link and a lte usb modem which runs with ncm protocol.
 
Any idea what I can do?

feckert, 2017-04-07:
@ThomasCr 

> My Problem is, that mwan3 hotplug script is not detecting, that my wwan connection is up. Only after running the follow command, it works:

Netifd will call all hotplug scripts in `/etc/hotplug.d/iface` if the interface is recognize as up by netifd. It could be possible that some parameters are not set during script call, and 15-mwan3 exits. This applies for example the gateway param. https://github.com/openwrt/packages/blob/master/net/mwan3/files/etc/hotplug.d/iface/15-mwan3#L43. Noticable is that after calling the script manual that all is working fine! Could you check that the netifd is calling all `/etc/hotplug.d/iface` scripts

feckert, 2017-11-03:
@ThomasCr ping?
could we close this issue?

feckert, 2018-03-15:
@hnyman i think we could close this no respond from the author since 11 months

ThomasCr, 2018-06-21:
yes, it can be closed.. but this ncm thing is not working well...

closed 2018-06-21 by ThomasCr