openwrt/packages #4239: freeradius3-democerts: The Demo Certs from the package freeradius3-democerts Expired on 3/Apr

loyukfai, 2017-04-06:
Maintainer: @tohojo 
Environment: LEDE 17.01.0

Description:
```
Validity
    Not Before: Feb  2 14:24:56 2017 GMT
    Not After : Apr  3 14:24:56 2017 GMT
```
It’s probably due to the makefile defaults has stipulated a valid day of 60 days.

Meanwhile, there's another issue of freeradius3 as described here:
https://bugs.lede-project.org/index.php?do=details&task_id=678

Cheers.

tohojo, 2017-04-08:
loyukfai <notifications@github.com> writes:

> Maintainer: @tohojo
> Environment: LEDE 17.01.0
>
> Description:
>
> Validity
>     Not Before: Feb  2 14:24:56 2017 GMT
>     Not After : Apr  3 14:24:56 2017 GMT
>
> It’s probably due to the makefile defaults [...]

I'll look them both over when I get a chance. Thanks for reporting :)

-Toke

loyukfai, 2017-04-18:
Welcome, and thanks for your follow-up in advance.

Cheers.

commit 424f4e2c (2017-04-26): freeradius3: A couple of small fixes

- Moves /etc/freeradius3/sites-{enabled,available}/inner-tunnel to be part of
  the freeradius3-mod-eap package. This prevents conflicts between
  freeradius3-mod-eap-peap and freeradius3-mod-eap-ttls which both included the
  file before. This fixes LEDE bug FS#678.

- Change the demo cert validity to be 1 year instead of 60 days. Should keep the
  cert valid for the duration of the LEDE release cycle (with some slack). This
  fixes #4239.
closed 2017-04-26 by tohojo