openwrt/packages #4242: packages/net/siit: siit.c doesn't compile.

wgrendel, 2017-04-09:
Maintainer: @Wedmer 
Environment: ipq806x, lede-17.01  

Description: siit.c doesn't compile.

```
*** siit.c.orig	2017-04-09 06:44:33.806461168 -0600
--- siit.c	2017-04-09 06:44:25.880389166 -0600
***************
*** 1166,1172 ****
  	siit_stats(dev)->rx_bytes += skb->len;
  	siit_stats(dev)->rx_packets++;
  
! 	dev->trans_start = jiffies;
  
  
  	/* Upper layer (IP) protocol forms sk_buff for outgoing packet
--- 1166,1172 ----
  	siit_stats(dev)->rx_bytes += skb->len;
  	siit_stats(dev)->rx_packets++;
  
!         netif_trans_update(dev);
  
  
  	/* Upper layer (IP) protocol forms sk_buff for outgoing packet

```

Wedmer, 2017-04-09:
Any compiler output? Kernel version?

wgrendel, 2017-04-09:
lede-17.01 target ipq806x.

> git remote show origin
> * remote origin
>   Fetch URL: git://git.lede-project.org/source.git
>   Push  URL: git://git.lede-project.org/source.git
>   HEAD branch: master
>   Remote branches:
>     lede-17.01 tracked
>     master     tracked
>   Local branch configured for 'git pull':
>     master merges with remote master
>   Local refs configured for 'git push':
>     lede-17.01 pushes to lede-17.01 (local out of date)
>     master     pushes to master     (up to date)

$ make V=s
make[4]: Entering directory 
'/home/XXX/Work/Lede/source/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/linux-4.9.20'
   CC [M] 
/home/wnreynolds/Work/Lede/source/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/siit-1.2/siit.o
/home/XXX/Work/Lede/source/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/siit-1.2/siit.c: 
In function 'siit_xmit':
/home/XXX/Work/Lede/source/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x/siit-1.2/siit.c:1169:5: 
error: 'struct net_device' has no member named 'trans_start'
   dev->trans_start = jiffies;
      ^

commit 43963958 (2017-04-10): [net/siit] Added compability for Linux >= 4.7.0
Closes #4242
closed 2017-04-10 by hnyman