openwrt/packages #4246: mwan3 -> on status luci page: => daemon.err uhttpd uci: Entry not found

camelcamro, 2017-04-10:
on current TRUNK, there is something wrong with the MWAN status information, as it is spaming the logs with:
Mon Apr 10 14:12:58 2017 daemon.err uhttpd[2754]: uci: Entry not found

see also:
https://dev.openwrt.org/ticket/19419

hnyman, 2017-04-10:
> on current TRUNK, there is something wrong with the MWAN

which exact trunk build?  Openwrt or LEDE and which git revision?

cc @feckert

camelcamro, 2017-04-10:
Sorry, forgot to tell:  Current lede trunk version

hnyman, 2017-04-10:
> Current lede trunk version

But which build of the LEDE trunk ???
And which mwan version in use? 
Mwan was updated just a few ago, so plase tell the exact version.
("opkg list-installed" or look at installed progs in LuCI)

That error "uci: Entry not found" has usually surfaced when there is an uninitialised or empty variable. Some config item has no default? and is missing from the config?

camelcamro, 2017-04-10:
Lede and mwan3 Version fromm today's build.
But maybe it is because had old settings. I will try to resetup from scratch

camelcamro, 2017-04-10:
ok, i was looking into ...
well, i have maybe a special case ...
as i have a WLAN - access point + client mode (hotspot to my handset android tethering AP).
THis is mostly disabled, as if not reachable, it would also bring down the AP mode of wireless lann.
so, this interface is most of time DOWN, and maybe this is the reason, that MWAN3 - is reporting that as empty config.
it is telling me:
WARNING: some interfaces are configured incorrectly or not at all in /etc/config/network!

this internface ("wwan") called in my case is in /etc/config/networks defined.
config interface 'wwan'
<------>option _orig_ifname 'wlan0'
<------>option _orig_bridge 'true'
<------>option proto 'dhcp'
<------>option delegate '0'
<------>option metric '40'
<------>option force_link '1'

whats wrong ?
why does MWAN3 not find it ?

but in my case, - if this interface is on - it would b valid, and it is then managed also correct.
so, that means, that this warning is not valid to report and spaming the logread. with: 

Mon Apr 10 19:56:17 2017 daemon.err uhttpd[2500]: uci: Entry not found

whats wrong ?

feckert, 2017-04-11:
@camelcamro the `uhttpd[xxx] uci: Entry not found` error is generated in this line https://github.com/openwrt/luci/blob/master/applications/luci-app-mwan3/luasrc/controller/mwan3.lua#L67 . It is related to the luci-app-mwan3 packages in the luci feed. Maybe we have to check if the interface is present in the config before we read a config option.

camelcamro, 2017-04-11:
sounds to be ... yes ...
but i have not that much insights about howto fix ..
if someone can try to fix it and build a package -> i can support you on testing

feckert, 2017-06-01:
I think we could close this

feckert, 2017-08-14:
@hnyman ping

closed 2017-08-14 by hnyman