openwrt/packages #4250: php7: php7-mod-gettext module is compiled without bind_textdomain_codeset function support

Dimazhan, 2017-04-11:
Environment: x86 generic, OpenWRT trunk

Description:

the php-mod-gettext module is compiled without bind_textdomain_codeset function support
the problem exists in chaos calmer and trunk
the script "configure" ignores transferred in parameter "-with-gettext" a way to the catalog "/usr/lib/libintl-full" and uses instead of him "/usr/lib/libintl-stub"
[config.log.gz](https://github.com/openwrt/packages/files/912633/config.log.gz)

error in browser
```
Fatal error: Uncaught Error: Call to undefined function bind_textdomain_codeset() in /www/zabbix/include/classes/core/ZBase.php:340 Stack trace: #0 /www/zabbix/include/classes/core/ZBase.php(145): ZBase->initLocales() #1 /www/zabbix/setup.php(27): ZBase->run('setup') #2 {main} thrown in /www/zabbix/include/classes/core/ZBase.php on line 340
```

Dimazhan, 2017-04-11:
I have corrected this error by means of this patch

[9000-gettext_fix.patch.txt](https://github.com/openwrt/packages/files/914086/9000-gettext_fix.patch.txt)

mhei, 2017-04-11:
The patch looks good to me, just waiting for a test compile before pushing the commit. We should try to get this patch upstream. There is a php codebase at [https://github.com/php/php-src](https://github.com/php/php-src). Do you want to open a PR there, or shall I do so?

mhei, 2017-04-11:
I've commited the patch but I'd like to leave this ticket open for me as a reminder until the upstream PR is created.

Dimazhan, 2017-04-12:
yes, of course. open you
I have found this problem in Openwrt Chaos Calmer php-5.6.17  
I thought in php7 already have corrected...)

mhei, 2017-04-13:
I've created an upstream ticket, so closing this here. Let's see whether upstream accepts the bugfix.

closed 2017-04-13 by mhei