openwrt/packages #4251: <Adblock>: 0 domains loaded successfully

Callumpy, 2017-04-11:
Apologies if this has already been posted, I just updated my router to LEDE with the latest version of Adblock, but according to the logs and the lack of ad blocking, the block lists aren't being downloaded.

`Tue Apr 11 12:44:15 2017 user.notice adblock-[2.5.1] info : start adblock processing ...
Tue Apr 11 12:44:15 2017 user.notice adblock-[2.5.1] debug: name: adaway, enabled: 1, backup: 1, dns: dnsmasq, fetch: /bin/uclient-fetch, memory: 28176
Tue Apr 11 12:44:15 2017 user.notice adblock-[2.5.1] debug: name: adaway, mode: restore, count: 0, in_rc: 1, out_rc: 1
Tue Apr 11 12:44:15 2017 user.notice adblock-[2.5.1] debug: name: adaway, mode: remove, count: 0, in_rc: 1, out_rc: 0
Tue Apr 11 12:44:15 2017 user.notice adblock-[2.5.1] debug: name: adguard, mode: remove, count: 0, in_rc: 0, out_rc: 0
Tue Apr 11 12:44:15 2017 user.notice adblock-[2.5.1] debug: name: blacklist, mode: remove, count: 0, in_rc: 0, out_rc: 0
Tue Apr 11 12:44:15 2017 user.notice adblock-[2.5.1] debug: name: disconnect, enabled: 1, backup: 1, dns: dnsmasq, fetch: /bin/uclient-fetch, memory: 28176
Tue Apr 11 12:44:15 2017 user.notice adblock-[2.5.1] debug: name: disconnect, mode: restore, count: 0, in_rc: 1, out_rc: 1
Tue Apr 11 12:44:16 2017 user.notice adblock-[2.5.1] debug: name: disconnect, mode: remove, count: 0, in_rc: 1, out_rc: 0
Tue Apr 11 12:44:16 2017 user.notice adblock-[2.5.1] debug: name: dshield, mode: remove, count: 0, in_rc: 0, out_rc: 0
Tue Apr 11 12:44:16 2017 user.notice adblock-[2.5.1] debug: name: feodo, mode: remove, count: 0, in_rc: 0, out_rc: 0
Tue Apr 11 12:44:16 2017 user.notice adblock-[2.5.1] debug: name: hphosts, mode: remove, count: 0, in_rc: 0, out_rc: 0
Tue Apr 11 12:44:16 2017 user.notice adblock-[2.5.1] debug: name: malware, mode: remove, count: 0, in_rc: 0, out_rc: 0
Tue Apr 11 12:44:16 2017 user.notice adblock-[2.5.1] debug: name: malwarelist, mode: remove, count: 0, in_rc: 0, out_rc: 0
Tue Apr 11 12:44:16 2017 user.notice adblock-[2.5.1] debug: name: openphish, mode: remove, count: 0, in_rc: 0, out_rc: 0
Tue Apr 11 12:44:16 2017 user.notice adblock-[2.5.1] debug: name: palevo, mode: remove, count: 0, in_rc: 0, out_rc: 0
Tue Apr 11 12:44:16 2017 user.notice adblock-[2.5.1] debug: name: ransomware, mode: remove, count: 0, in_rc: 0, out_rc: 0
Tue Apr 11 12:44:17 2017 user.notice adblock-[2.5.1] debug: name: reg_ch, mode: remove, count: 0, in_rc: 0, out_rc: 0
Tue Apr 11 12:44:17 2017 user.notice adblock-[2.5.1] debug: name: reg_pl, mode: remove, count: 0, in_rc: 0, out_rc: 0
Tue Apr 11 12:44:17 2017 user.notice adblock-[2.5.1] debug: name: reg_ro, mode: remove, count: 0, in_rc: 0, out_rc: 0
Tue Apr 11 12:44:17 2017 user.notice adblock-[2.5.1] debug: name: reg_ru, mode: remove, count: 0, in_rc: 0, out_rc: 0
Tue Apr 11 12:44:17 2017 user.notice adblock-[2.5.1] debug: name: securemecca, mode: remove, count: 0, in_rc: 0, out_rc: 0
Tue Apr 11 12:44:17 2017 user.notice adblock-[2.5.1] debug: name: shalla, mode: remove, count: 0, in_rc: 0, out_rc: 0
Tue Apr 11 12:44:17 2017 user.notice adblock-[2.5.1] debug: name: spam404, mode: remove, count: 0, in_rc: 0, out_rc: 0
Tue Apr 11 12:44:17 2017 user.notice adblock-[2.5.1] debug: name: sysctl, mode: remove, count: 0, in_rc: 0, out_rc: 0
Tue Apr 11 12:44:17 2017 user.notice adblock-[2.5.1] debug: name: whocares, mode: remove, count: 0, in_rc: 0, out_rc: 0
Tue Apr 11 12:44:17 2017 user.notice adblock-[2.5.1] debug: name: winspy, mode: remove, count: 0, in_rc: 0, out_rc: 0
Tue Apr 11 12:44:17 2017 user.notice adblock-[2.5.1] debug: name: winhelp, mode: remove, count: 0, in_rc: 0, out_rc: 0
Tue Apr 11 12:44:17 2017 user.notice adblock-[2.5.1] debug: name: yoyo, enabled: 1, backup: 1, dns: dnsmasq, fetch: /bin/uclient-fetch, memory: 28176
Tue Apr 11 12:44:18 2017 user.notice adblock-[2.5.1] debug: name: yoyo, mode: restore, count: 0, in_rc: 1, out_rc: 1
Tue Apr 11 12:44:18 2017 user.notice adblock-[2.5.1] debug: name: yoyo, mode: remove, count: 0, in_rc: 1, out_rc: 0
Tue Apr 11 12:44:18 2017 user.notice adblock-[2.5.1] debug: name: zeus, mode: remove, count: 0, in_rc: 0, out_rc: 0
Tue Apr 11 12:44:29 2017 user.notice adblock-[2.5.1] info : block lists with overall 0 domains loaded successfully (LEDE Reboot 17.01.0 r3205-59508e3)`

Help would be appreciated.

hnyman, 2017-04-11:
@dibdot

Callumpy, 2017-04-11:
Ok, having checked through the code and the config files, I realized it's because wget with SSL support is not installed. By default wget-nossl is installed on the firmware.

Can the downloading of block lists be done without the need for SSL to cater for users of low memory devices?

dibdot, 2017-04-11:
adblock tries to use 'uclient-fetch' but you're missing 'libustream-mbedtls' ... please install this library as well - no need for wget plus openssl.

hnyman, 2017-04-11:
@dibdot 
is is possible to download only those blocklists that use http?
(user can see from /etc/config/adblock, which blocklists are using the plain http URLs)

dibdot, 2017-04-11:
@hnyman 
In forthcoming 2.6.0 release I've added an additional check for '/lib/libustream-ssl.so' so that the user gets a clear error message. Sadly nearly all "http only" sites are providing very big lists with > 10.000 domains ... not useful for low memory devices.

Callumpy, 2017-04-11:
Hey @dibdot.

I use just the three lists that are selected by default, adaway, disconnect, and yoyo. I really have no idea which ones are the best, so just trusted you chose the best ones for me :)

All three are available without the use of SSL/TLS.
> http://adaway.org/hosts.txt
> http://s3.amazonaws.com/lists.disconnect.me/simple_malvertising.txt
> http://pgl.yoyo.org/adservers/serverlist.php?hostformat=nohtml&showintro=0&mimetype=plaintext

Would it be worth giving the option to load them over a secure connection? I personally don't care if they aren't. If it means I get to install less things on my old router, then it seems like a win to me. 

All the best!

dibdot, 2017-04-12:
@Callumpy adaway is a SSL only site. Anyway, currently I'll prepare limited "http-only" support (wget-nossl / uclient-fetch without libustream-ssl) for the next release 2.6 - stay tuned!

Callumpy, 2017-04-12:
Sounds good @dibdot. Adaway is SSL only after-all, looks like some of the remains of wget are still on my router when I tested it. I don't actually remember what dependencies it installed!

Anyway, I'll look forward to your updates! Is there a way I can follow your updates? Watching the repo on git watches the entire packages repo of course.

dibdot, 2017-04-12:
@Callumpy  see #4253 , if you would like to use more http sites, please change the source urls in your config manually, default config will use https if possible.

br
dirk

hnyman, 2017-04-12:
http download support in adblock should work since #4253 got merged. 
(but naturally only for those sites that offer blacklists via http )

closed 2017-04-12 by hnyman