openwrt/packages #4252: linux-4.9.20 - x86_64 cyrptographical hardware modules not being added to images

aenertia, 2017-04-11:
Maintainer: ndb@nbd.name
Environment: x86_64, Ubuntu 17.04 Build Host, lede/master commit: f4e312ddf855375667f43c842c7187934b8aca92

Description:

Cyrptographic hardware modules are not built by default including intel aes-ni; depsite doing a make kernel_menuconfig they appear to get stripped/removed during a full image build operation. The modules do get compiled but are never added to images. Adding them into the kernel does not work as intended either. 

```
clone from lede project;
make menuconfig
select x86_64, enable openssl build options to support dynamic module loading
make kernel_menuconfig 
select appropriate kernel modules for the platform (used i7-7700k as target)
make 
load combined root image into kvm/qemu with host-passthrough on.
Check dmesg and lsmod for kernel modules. None found. 
run openssl engine - check output. 
Run openssl with accelerated ciphers and check time against host-os with aes-ni - significantly slower in lede

```
This will impact a lot of build targets; including newer atom boards used in embedded routers/firewalls.

aenertia, 2017-04-12:
Attached .config and kernel .config from a build (note this is glibc build with gcc with uclibc and musl is the same) 
[lede.config.txt](https://github.com/openwrt/packages/files/915213/lede.config.txt)
[kernel.config.txt](https://github.com/openwrt/packages/files/915217/kernel.config.txt)

nbd168, 2017-04-12:
AES-NI support is built into the kernel directly, not as a module. Did you install kmod-cryptodev in your image?

nbd168, 2017-04-12:
By the way, this is the wrong place for a LEDE bug report. Use this: https://bugs.lede-project.org/

closed 2017-04-12 by nbd168

aenertia, 2017-04-12:
Yes. Built in but it isn't getting exposed properly.

nbd168, 2017-04-12:
The config you posted has kmod-cryptodev selected as 'm', so not built in. Did you change it in the mean time?

aenertia, 2017-04-12:
That cryptodev gets loaded and I can get openssl to show it as available
(BSD cryptodev) the problem comes when you run ciphers performance looks
very much like unaccelerated perf numbers. It could just be VM overhead but
I don't get that behaviour on my full fledged Linux VM's (Ubuntu/fedora e.g)

nbd168, 2017-04-12:
You forgot to enable these options:
# CONFIG_OPENSSL_HARDWARE_SUPPORT is not set
# CONFIG_OPENSSL_ENGINE_CRYPTO is not set