openwrt/packages #4255: net/vnstat: vnstati package cannot be built anymore

xdarklight, 2017-04-12:
Maintainer: @diizzyy @jow- 
Environment: LEDE git master lede-project/source@ea1855949b

Description:
enable CONFIG_PACKAGE_vnstati (in my case on the lantiq xRX200 target)

```
$ grep vnstat .config | grep "="
CONFIG_PACKAGE_luci-app-vnstat=y
CONFIG_PACKAGE_vnstat=y
CONFIG_PACKAGE_vnstati=y
```
```
...
configure: 
configure:  ============================================== 
configure: 
configure:          host: mips-openwrt-linux-gnu 
configure:         build: x86_64-pc-linux-gnu 
configure: 
configure:        prefix: /usr     
configure:    sysconfdir: /etc 
configure: 
configure:    Install paths 
configure: 
configure:     vnstat: /usr/bin/vnstat 
configure:    vnstatd: /usr/sbin/vnstatd 
configure:     config: /etc/vnstat.conf 
configure: 
configure:           Tests: unavailable 
configure:    Image output: disabled 
configure: 
configure:  ============================================== 
configure: 
...
install -d -m0755 lede/build_dir/target-mips_24kc_musl/vnstat-1.17/ipkg-mips_24kc/vnstati/usr/bin
cp -fpR lede/build_dir/target-mips_24kc_musl/vnstat-1.17/ipkg-install/usr/bin/vnstati lede/build_dir/target-mips_24kc_musl/vnstat-1.17/ipkg-mips_24kc/vnstati/usr/bin/
cp: cannot stat 'lede/build_dir/target-mips_24kc_musl/vnstat-1.17/ipkg-install/usr/bin/vnstati': No such file or directory
make[3]: *** [Makefile:117: lede/bin/packages/mips_24kc/packages/vnstati_1.17-1_mips_24kc.ipk] Error 1
...
```

This seems to be caused by upstream commit /vergoh/vnstat@8a4a9b82d4f8fd39db8417fd06d969505fde9d0a

xdarklight, 2017-04-12:
before I forget:
```
$ grep libgd .config
CONFIG_PACKAGE_libgd=y
# CONFIG_PACKAGE_libgdbm is not set
```

I think this issue is caused by b8cf63a654e86d7e9e40bff7eaf276ae364dc459 (but I haven't done git bisect yet)

diizzyy, 2017-04-12:
Works fine here, are you using a clean tree?
The buildbots are building it fine too.
http://downloads.lede-project.org/snapshots/packages/mipsel_24kc/packages/

xdarklight, 2017-04-12:
indeed, removing build_dir (and staging_dir, just to be sure...) fixed this.
thanks for the pointer though!

closed 2017-04-12 by xdarklight