openwrt/packages #4257: <packages/net/openssh/> Query about key generation

anithacp, 2017-04-13:
We are trying to integrate sysrepo, netopeer server, libyang, libnetconf2 into an exisiting Openwrt build. Our binary image size was around 5.4 MB before the integration and now has increased to around 6.8 MB post integration with PMA modules including openssh. Currently dropbear to login to the device. Since netopeer server requires openssh package,we have the following questions

1. Can the existing dropbear keys be used instead of openssh? I tried the following and wasn’t successful.
```
root@XXX:/# dropbearkey -t rsa -f id_rsa
Generating key, this may take a while...
Public key portion is:
root@XXX:/# dropbearkey -y -f id_rsa | grep "^ssh-rsa " >> authorized_keys
root@XXX:/# cp authorized_keys /etc/keystored/keys/ssh_host_rsa_key.pub.pem
root@XXX:/# sysrepocfg -d startup -i /etc/sysrepo/xml/stock_key_config.xml ietf-keystore
The new configuration was successfully applied.
```
However the client connection to server failed
```
 # ./cli/build/netopeer2-cli
> connect --ssh --port 830 --login root --host 192.168.1.1
nc ERROR: Starting the SSH session failed (Socket error: Connection reset by peer).
cmd_connect: Connecting to the 192.168.1.1:830 as user "root" failed.
```
2. Private key generated by SSH is used as is while openssl is used to generate the public key from the SSH public key. Why is this needed? Is there a way to use the SSH public key directly without requiring the openssl package?
```
root@XXX:/# ls -l /etc/ssh/
-rw-r--r--    1 root     root        246880 Apr 11 04:11 moduli
-rw-------    1 root     root           227 Apr 11 04:37 ssh_host_ecdsa_key
-rw-r--r--    1 root     root           175 Apr 11 04:37 ssh_host_ecdsa_key.pub
-rw-------    1 root     root           399 Apr 11 04:37 ssh_host_ed25519_key
-rw-r--r--    1 root     root            95 Apr 11 04:37 ssh_host_ed25519_key.pub
-rw-------    1 root     root          1679 Apr 11 04:37 ssh_host_rsa_key
-rw-r--r--    1 root     root           395 Apr 11 04:37 ssh_host_rsa_key.pub
-rw-r--r--    1 root     root          3640 Apr 11 04:11 sshd_config
root@XXX:/# cp /etc/ssh/ssh_host_rsa_key.pub /etc/keystored/keys/ssh_host_rsa_key.pub.pem
```
Above also cause connection failure.

Above questions arises mainly because we have a size limitation on the embedded device. We want to limit the image size to less than 6.5MB.

Thanks and regards,
Anitha

anithacp, 2017-04-13:
Just to add to it: We are using development branches for all the modules,  updated to latest as of April 10, 2017.

neheb, 2019-03-30:
@anithacp is this still an issue? dropbear has received many updates.

anithacp, 2019-04-02:
@neheb This is not an issue anymore. Thanks

closed 2019-04-02 by neheb